dangowrt/owrt-ubi-installer #114: 5G Doesn´t work with Signed Build V 0.65

ArcticHawk1, 2022-08-21:
After I update into OpenWrt my router can´t do updates because of kernel incompatibility and it doesn´t see my 5G radio.

mrkiko, 2022-08-21:
>     0.65 (Issue #114)

Hello!!
Currently, I do not have a clear idea of the state of your device, hence I am guessing here;
I guess you mean you converted your device to UBI layout, by flashing the file with the word "installer" in the name, so you switched from stock firmware (+ stock bootloader) to UBI (Daniel OpenWRt bootloader + ubi layout).

Now, two scenarios are possible:
1 - You might be in recovery mode: if I remember correctly, mt7615 (5 GHZ) driver is not present, hence your 5G radio is not detected.
In this case, maybe have a cursory look at /sys/fs/pstore, deleting whaetever is there if anything is present, and thn proceeding with sysupgrade:
sysupgrade -n <imagename>
you may obtain a swuitable image from here:
https://downloads.openwrt.org/releases/22.03.0-rc6/targets/mediatek/mt7622/openwrt-22.03.0-rc6-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

2 - You might have flashed a sysupgrade image, but the device may still not reboot in the Production system.
Have a cursory look at /sys/fs/pstore and remove whatever is inside that folder, if anything is present, then reboot.
If all goes well, you might now be in the production system, which should contain the mt7615 driver, so that your 5 GHZ radio should be detected.

Then there are other possibilities, but without more informations it's difficult to say.
Please let me know more, and I will try my best to help out.

ArcticHawk1, 2022-08-21:
Thank you a lot. Issue is fixed after doing the update manually from the console instead of luci.

closed 2022-08-21 by ArcticHawk1