dangowrt/owrt-ubi-installer #116: The image file is incorrect

zwwuu, 2022-09-08:
On a Linksys E8450 with firmware version 1.2.00.273012, and uploading [openwrt-22.03.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.0.0/openwrt-22.03.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb) results in popup sating "The image file is incorrect".

dangowrt, 2022-09-12:
That's odd. I've tried using the installer after flasing FW_E8450_1.2.00.273012_PROD_signed.img and that worked for me. I did not try flashing the recovery image from the stock loader though...
Does using [openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v0.6.5/openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb) just for the backup step work?

zwwuu, 2022-09-12:
No, same popup "The image file is incorrect".

dangowrt, 2022-09-12:
Which browser are you using? Can you post a screenshot of the firmware version information showing in the web-ui?

zwwuu, 2022-09-12:
Firefox 104.0.2 on Windows 10. I also tried on Chrome and Edge, same popup for all browsers. [https://i.imgur.com/UsMzGgF.png](screenshot)

dangowrt, 2022-09-12:
Very strange indeed. Maybe just try if the installer itself works (no need for those very complete backups: the installer will also do a backup of the previous bootchain just missing the firmware file itself which can easily be downloaded from Linksys/Belkin).

dangowrt, 2022-09-12:
Ha! I have vague memories of there being a problem if the complete path (something like `C:\....\openwrt-foo-bar.itb` on Windows) is too long/comlex. Try storing the file in a very simple path, like the root directory of a drive.

zwwuu, 2022-09-12:
I renamed installer image to `installer.itb` and placed at the root of C drive. Same popup again.

dangowrt, 2022-09-13:
Please confirm you are exactly on `1.2.00.273012` vendor firmware version.

zwwuu, 2022-09-13:
Yes
[image]

dangowrt, 2022-09-14:
I've retried with my device, went all the way back to stock firmware 1.00, did the two automatic updates on the way to 1.2.00.273012. When ahead and downloaded the installer file from github, uploaded to router and all works. I'm really a bit puzzled about what the cause of your difficulties may be.

zwwuu, 2022-10-13:
@wallrik No, even with `.img` extension i still get error. 
[image]

dangowrt, 2022-10-13:
Does flashing the original vendor firmware work?

wallrik, 2022-10-13:
Okay, long shot - What if, just what if... Have you been using the same downloaded file for every attempt? What if it was corrupted for some weird reason...

```powershell
PS C:\Users\wallrik\Downloads> Get-FileHash "openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb"

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          8392F88DA4A50441DA1642C9D22445099787E810890CA148D56109AE2FE0FBDD       C:\Users\wallrik\Downloads\openwrt-22.03.1-me...
```

zwwuu, 2022-10-14:
@dangowrt No, same error with original firmware `FW_E8450_1.2.00.273012_PROD_signed.img`.

dangowrt, 2022-10-15:
In case you haven't done that yet, try doing a factory reset to erase all settings before flashing. If that also doesn't help and you can't even flash the original firmware, then there is most likely another problem with the device and you will have to decide at this point if you just want to return it (in case you got warranty) or open it, connect the serial port and find out what's actually happening (which implies loosing the warranty).

rthomas67, 2022-11-15:
# Summary
Encountered the same  "The image file is incorrect" issue trying to flash the recovery image (not installer) starting from stock firmware 1.1.00.180912

# TLDR
* **Tried (AFTER trying many other things) flashing the image that is not suffixed with _signed and that worked fine.** Backup process complete via SSH.
* Instructions in the README for "complete backup" say to **"use _signed.itb variant if running stock firmware >= 1.1"** but that is not quite right.  Apparently some of the 1.1 firmware builds should still use the unsigned images.
  * @dangowrt In addition to the request from @rvernica in issue #122 to update the README to say the "complete backup" isn't really necessary, and the automatic "minimal backup" made during UBI conversion is sufficient, please **also** update the README to clarify the exact 1.1 stock firmware build version after which the _signed.itb file should be used (data point -> build 1.1.00.180912 is apparently **before** the _signed image is needed).  Also, probably remove the section about using Luci to create the backup mtd files (or at least add a warning about it) since there are reports that those files differ from the ones produced by the SSH / dd / SCP based process (maybe Luci just base64 encodes the data or something like that, but the raw dd files seem to be a "cleaner" option).

# Context
(This is a summary of some of the "wheel-spinning" caused by the slightly incorrect statement about stock firmware version 1.1 in the README.)
* Renaming the file from .itb to .img doesn't work.  Generally the server end of a browser file upload (the firmware image in this case) doesn't care what the file is named on the client side, and this appears to hold true here too.
  * Renaming the client/brower side file to a short file name / short path doesn't change anything either.
* Used web browser Chromium on Raspberry Pi OS (and Raspberry Pi 4B hardware) - Trying to eliminate client platform / browser as a possible contributing cause... but this was also not a factor.
* When the stock Linksys e8450 web UI shows the "...incorrect..." popup, message, the browser console log (developer tools) shows this message (maybe this is the raw response from the http request to config-admin-firmware.html): `{result: "Invalid firmware format."}` followed by another message: `fail`
  * There was no indication in the dev tools console of the jQuery fetch error posted by @wallrik .  This confirmed that the entire firmware image was being successfully sent to the e8450 and validated somehow.
* Based on https://www.linksys.com/support-article?articleNum=317332 having only one "hardware version" to choose from (1.0), there is no reason to conclude that this issue is related to hardware differences in the e8450.
* `sha256sum openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb` matches the one posted above. **Note that the verified checksum is for the -installer_signed.itb file and not the -recovery_signed.itb one mentioned by the OP** but transitively, there doesn't seem to be a download integrity/corruption issue here.
* Other circumstantial stuff that could have been a factor:
  * The e8450 was **not** connected to anything on the WAN side / no internet access during firmware upgrade attempt.
    * This raised suspicion that there may have been some dependency within the stock firmware on an external resource (library fetch, call to "mother ship", etc.) 
  * The e8450 "setup wizard" was aborted (because it could not complete w/o a WAN connection).  i.e. no WAN IP assignment, no auto-updates from stock, root p/w still defaulted to 'admin', etc. 
    * This could have been something that affected the "upgrade" logic in the e8450, if it checked the status of something that should be "secure" before allowing an upgrade.

fiveangle, 2022-12-20:
According to [Linksys release notes](https://downloads.linksys.com/support/assets/releasenotes/E8450_1.2.00.273012_Customer_Release_Note.txt), support for signed FW was introduced in 1.1.01.272918:  

```
---------------------------------------------------------------------------
Firmware version:   1.1.01.272918
Release date:       April 13, 2022

- Supported future signed image for next firmware release. 
---------------------------------------------------------------------------
```
but not enforced until 1.2.00.273012:
```
---------------------------------------------------------------------------
Firmware version:   1.2.00.273012
Release date:       April 13, 2022

- Add support for signed firmware image only. Once firmware is updated, you can no longer rollback to a previous unsigned firmware image. 
…
---------------------------------------------------------------------------

```

So instructions everywhere should be changed to: ">=1.2.00.273012"

dangowrt, 2022-12-21:
> So instructions everywhere should be changed to using signed FW image only for versions >=1.2.00.273012

Can you make a pull-request for README.md for the change you are suggesting?

fiveangle, 2022-12-22:
> > So instructions everywhere should be changed to using signed FW image only for versions >=1.2.00.273012
> 
> Can you make a pull-request for README.md for the change you are suggesting?

#130

PR #130 (2022-12-31): Fix documentation which firmware versions require signed vs. unsigned
Per [Linksys release notes](https://downloads.linksys.com/support/assets/releasenotes/E8450_1.2.00.273012_Customer_Release_Note.txt
) firmware signing requirement is enforced only for versions >= 1.2.00.273012

Resolves #116
closed 2022-12-31 by dangowrt