dangowrt/owrt-ubi-installer #122: "factory reset" step in "Device flash complete backup procedure" could brick the device?

yuko1225, 2022-10-26:
I followed the instructions of the [Device flash complete backup procedure] in README. It went well until the final factory reset step. 

> 4. When the device is running original/vendor firmware, perform a factory reset.

After I did the factory reset, the device never boot back. Somehow the factory reset corrupt the vendor firmware and I cannot boot into it anymore. I tried the "reboot-torture" method as someone suggested, and it is either boot into the initramfs recovery, or not booting at all(no response from ARP nor ping in 90 seconds and keeps blinking white light).

The OpenWrt in the initramfs mode is still usable. However it is dangerous to flash anything on that recovery as the instructions cleary stated. To recover from it I had to restore the vendor's backup using the procedure in [Restoring the vendor/official firmware], which carries a risk. Luckily for me there is no bad-block on the first 22M of the flash.

I hope I can help others to avoid similar situation. I can provide the corrupted mtd3 data if necessary.

dangowrt, 2022-10-30:
Yet another reason to *not* carry out that full backup and rely on the minimal backup which the installer does automatically. The minimal backup is sufficient to restore the device to stock firmware, as you can download the firmware file itself from the vendor's website...

yuko1225, 2022-11-01:
> Yet another reason to _not_ carry out that full backup and rely on the minimal backup which the installer does automatically. The minimal backup is sufficient to restore the device to stock firmware, as you can download the firmware file itself from the vendor's website...

@dangowrt But the README said otherwise:
> To be 100% on the safe side consider doing the [complete backup procedure](https://github.com/dangowrt/owrt-ubi-installer#device-flash-complete-backup-procedure) before proceeding with the installation.

rvernica, 2022-11-06:
How did you backup the mtdblock contents? I did both luci and ssh and the file sizes are way different:
```
./luci:
total 1424
-rw-r--r--. 1    524288 Nov  6 09:57 OpenWrt.mtd0.bin
-rw-r--r--. 1    802816 Nov  6 09:57 OpenWrt.mtd1.bin
-rw-r--r--. 1         0 Nov  6 10:08 OpenWrt.mtd2.bin
-rw-r--r--. 1    131072 Nov  6 09:57 OpenWrt.mtd3.bin

./ssh:
total 130816
-rw-r--r--. 1    524288 Nov  6 10:04 mtd0
-rw-r--r--. 1   1310720 Nov  6 10:04 mtd1
-rw-r--r--. 1   1048576 Nov  6 10:04 mtd2
-rw-r--r--. 1 131072000 Nov  6 10:04 mtd3
```

yuko1225, 2022-11-06:
> How did you backup the mtdblock contents? I did both luci and ssh and the file sizes are way different:
>
> ```
> ./luci:
> total 1424
> -rw-r--r--. 1    524288 Nov  6 09:57 OpenWrt.mtd0.bin
> -rw-r--r--. 1 [...]

@rvernica The luci method is known defected. Use SSH method only.

rvernica, 2022-11-06:
> Yet another reason to _not_ carry out that full backup and rely on the minimal backup which the installer does automatically. The minimal backup is sufficient to restore the device to stock firmware, as you can download the firmware file itself from the vendor's website...

BTW, this might be useful to add to the README. It was not clear that the minimal backup is sufficient to restore the stock fw.

dangowrt, 2022-12-01:
> BTW, this might be useful to add to the README. It was not clear that the minimal backup is sufficient to restore the stock fw.

Done by commit https://github.com/dangowrt/owrt-ubi-installer/commit/b2a8d6032ce67a2257deb9cfa2124d31d778ef4f

closed 2025-07-04 by dangowrt