dangowrt/owrt-ubi-installer #123: UART TTL Console need to login

xqy281, 2022-10-27:
Hi @dangowrt & @jauderho :bowtie:

In [FW_E8450_1.1.00.180912_PROD_signed.img ] or [FW_E8450_1.2.00.273012_PROD_signed.img ] using TTL  console degub is need to password login.

I try to using john(John the Ripper) crack /etc/shadow『$1$SxrX92zz$GC0W8uWvY/CnbP9fJM6Vd/』,But it's so hard:disappointed:

dangowrt, 2022-10-29:
The password of the `root` user if what you can set in the Web-UI for user `admin`. No need to crack it ;)

xqy281, 2022-10-31:
> `root` user if what you can set in the Web-UI for user `admin`. No need to crack it ;)

In Linksys Web-UI, how to set root password by user admin?

3Q~

dangowrt, 2022-10-31:
The wizard when first connecting to the device Web-UI actually forces you to set a password for the admin login on the Web-UI. **That very same password is then also the password for the `root` user**

xqy281, 2022-11-02:
> 

I tried to login as root user with wizard admin password but it still failed.
[image]
[image]