dangowrt/owrt-ubi-installer #124: ubi-squashfs-sysupgrade is copied without additions or subtractions

clutch2sft, 2022-10-31:
Maybe this was intentional? 

Running build_installer.sh results in five files:

openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb
openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb
openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb
openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

In the event where I only need openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb because I am upgrading from a different release (say 22.03.0) this image doesn't contain anything I added or subtracted with:

OPENWRT_REMOVE_PACKAGES
OPENWRT_ADD_PACKAGES

It seems the script is working as designed so I am not clear if this was a miss or intentional?

Thank you,

ggg

dangowrt, 2022-10-31:
Yes, this is intentional. In the beginning, when there were only snapshot images available, I wanted to enable common users to try OpenWrt on this device, hence it made sense to bundle LuCI (which is not included in snapshot images), and I used that opportunity to also include other packages I thought would be useful.
Now that there is the 22.03 release which already contains LuCI I decided to phase out the modified sysupgrade image in favor of using the officially released one. To me that makes more sense as then it's clear that you are just using official OpenWrt, and should hence also discuss or ask for support [in the official OpenWrt channels](https://openwrt.org/contact).

Also, it's easy for users to generate custom images with their individual selection of additional packages added using the [OpenWrt Firmware Selector](https://firmware-selector.openwrt.org/?target=mediatek%2Fmt7622&id=linksys_e8450-ubi).

clutch2sft, 2022-10-31:
Thanks for a speedy resonse and the references!  I hadn't yet stumbled across that customizable firmware selector.  Cool stuff!

closed 2022-10-31 by clutch2sft