dangowrt/owrt-ubi-installer #126: Link for Belkin FW is dead

4r1-60ld, 2022-11-21:
Don't know if this is the correct place for it, but it appear the link for the downgrade Belkin RT 3200 is dead and I cannot find it.... 
Thank you.

mdr92, 2022-11-22:
I hit the same problem, here's the new link, just s3 instead of www, the firmware is probably interchangeable but the Linksus has more lights if that would make a difference.
https://s3.belkin.com/support/assets/belkin/firmware/FW_RT3200_1.0.01.101415_prod.img

4r1-60ld, 2022-11-26:
> I hit the same problem, here's the new link, just s3 instead of www, the firmware is probably interchangeable but the Linksys has more lights if that would make a difference.
> https://s3.belkin.com/support/assets/belkin/firmware/FW_RT3200_1.0.01.101415_prod.img

Thank you!

commit a39d35c4 (2022-12-01): README.md: fix link to Belkin stock firmware

The URL of the stock firmware has changed.

Fixes: #126
closed 2022-12-01 by dangowrt