dangowrt/owrt-ubi-installer #127: Can't exit "recovery (initramfs) mode"

leonidas-1, 2022-11-27:
Hello!
I'm about to flash OpenWRT on my Belkin RT3200 and I'm following your docs in order to save a backup before. 

I flashed my router with the `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb` installer and downloaded the required files.

I'm now trying to revert to stock firmware (1.0.x) by rebooting (powercycling) the router as you describe in the guide but any attempt goes wrong (via ssh too). My router just reboot on OpenWRT Luci page and warn me that I'm in recovery mode.

[image]

How have I to proceed in order to avoid brick? 

Thank you!

leonidas-1, 2022-11-27:
I've just realized that I installed `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb` instead of `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb` as you warned about.

I'm an idiot.

What can I do now?

Many thanks!

dangowrt, 2022-11-27:
Don't worry, this is not a problem. Please verify that you are now running the new bootchain, just to be sure, do in SSH console:
```
fw_printenv ver
```
This should show
```
ver=U-Boot 2021.10 (Oct 16 2021 - 01:00:47 +0000)
```

If this is the case, then you can go ahead and flash the sysupgrade image.

leonidas-1, 2022-11-27:
Hello @dangowrt thank you for the support and for the big work you're doing!

Here my output:
`root@OpenWrt:~# fw_printenv ver`
`ver=U-Boot 2022.01-OpenWrt-r19777-2853b6d652 (Oct 07 2022 - 23:34:56 +0000)`
Is it dangerous to proceed with this bootchain?

Another question:
Are that files I've downloaded still good for an eventually backup in the future?

Thanks!

dangowrt, 2022-11-27:
> Is it dangerous to proceed with this bootchain?
As you have already run the installer (and that worked fine, that's why you see `U-Boot 2022.01` there now) you can safely proceed and flash the sysupgrade image.

> Are that files I've downloaded still good for an eventually backup in the future?

No, what you have downloaded now contains the already replaced bootchain and will not help you to revert to stock firmware. However, the installer also creates backup files which are sufficient to revert to stock, you'll just have to download the actual firmware from Belkin/Linksys as explained [here](https://github.com/dangowrt/owrt-ubi-installer#restoring-the-vendorofficial-firmware).

leonidas-1, 2022-11-27:
Ok I got it!
I've successfully flashed the sysupgrade image and I'm closing the issue.

Thank you!!

closed 2022-11-27 by leonidas-1