dangowrt/owrt-ubi-installer #128: initramfs recovery

pthomas, 2022-12-04:
Hello, my Belkin RT3200 w/ openwrt has been working great for the last year, but now I've gotten it into a semi-broken state. It boots to recovery mode (initramfs) fine, but I don't think everything is correct with the partitions. I've done the:
`rm /sys/fs/pstore/*` but it still doesn't do the full boot.

I can see what the layout is supposed to be here:
https://openwrt.org/toh/linksys/e8450#flash_layout

And that structure is what I have:
`[    1.759575] Creating 4 MTD partitions on "1100d000.snfi":
[    1.764972] 0x000000000000-0x000000080000 : "bl2"
[    1.770614] 0x000000080000-0x0000001c0000 : "fip"
[    1.777042] 0x0000001c0000-0x0000002c0000 : "factory"
[    1.783465] 0x000000300000-0x000008000000 : "ubi"u`

`root@OpenWrt:~# ubinfo -d 0 -a
ubi0
Volumes count:                           7
Logical eraseblock size:                 126976 bytes, 124.0 KiB
Total amount of logical eraseblocks:     1000 (126976000 bytes, 121.0 MiB)
Amount of available logical eraseblocks: 0 (0 bytes)
Maximum count of volumes                 128
Count of bad physical eraseblocks:       0
Count of reserved physical eraseblocks:  20
Current maximum erase counter value:     3
Minimum input/output unit size:          2048 bytes
Character device major/minor:            248:0
Present volumes:                         0, 1, 2, 3, 4, 5, 6

Volume ID:   0 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        9 LEBs (1142784 bytes, 1.0 MiB)
State:       OK
Name:        ubootenv
Character device major/minor: 248:1
-----------------------------------
Volume ID:   1 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        9 LEBs (1142784 bytes, 1.0 MiB)
State:       OK
Name:        ubootenv2
Character device major/minor: 248:2
-----------------------------------
Volume ID:   2 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        72 LEBs (9142272 bytes, 8.7 MiB)
State:       OK
Name:        recovery
Character device major/minor: 248:3
-----------------------------------
Volume ID:   3 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        67 LEBs (8507392 bytes, 8.1 MiB)
State:       OK
Name:        boot_backup
Character device major/minor: 248:4
-----------------------------------
Volume ID:   4 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        30 LEBs (3809280 bytes, 3.6 MiB)
State:       OK
Name:        fit
Character device major/minor: 248:5
-----------------------------------
Volume ID:   5 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        40 LEBs (5079040 bytes, 4.8 MiB)
State:       OK
Name:        rootfs
Character device major/minor: 248:6
-----------------------------------
Volume ID:   6 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        749 LEBs (95105024 bytes, 90.6 MiB)
State:       OK
Name:        rootfs_data
Character device major/minor: 248:7`

And I see what the FIT startup is supposed to look like from the boot log (https://openwrt.org/toh/linksys/e8450#openwrt_bootlog):
`[    1.810153] FIT: volume size: 21576 sectors (11046912 bytes)
[    1.815823] FIT: FDT structure size: 4096 bytes
[    1.820398] FIT: Default configuration: config@1 (OpenWrt linksys_e8450-ubi)
[    1.827456] FIT:           kernel sub-image 0x00001000 - 0x004b2f91 'kernel@1' (ARM64 OpenWrt Linux-5.10.18) 
[    1.837376] FIT:          flat_dt sub-image 0x004b3000 - 0x004b8b84 'fdt@1' (ARM64 OpenWrt linksys_e8450-ubi device tree blob) 
[    1.848859] FIT:       filesystem sub-image 0x004b9000 - 0x00a84000 'rootfs@1' (ARM64 OpenWrt linksys_e8450-ubi rootfs) 
[    1.859726] FIT: selecting configured loadable rootfs@1 to be root filesystem
[    1.866857]  ubiblock0_4: p1(rootfs@1)`
But I don't see this in my dmesg.

Now for the questions. Are there Linux commands to inspect & mount the FIT partition? Can I directly write the ubi0_4 fit image using `mtd write`? What image is that?

Any help here would be appreciated. Thanks again for the great ubi installer!

pthomas, 2022-12-04:
OK, I removed `rootfs` volume.

'/sys/fs/pstore/dmesg-ramoops-0' is here:
[dmesg-ramoops-0.txt](https://github.com/dangowrt/owrt-ubi-installer/files/10149287/dmesg-ramoops-0.txt)

Thanks for the help, I really appreciate it!

dangowrt, 2022-12-04:
```
<5>[    0.964258] 0x000000000000-0x000000080000 : "Preloader"
<5>[    0.970407] 0x000000080000-0x0000000c0000 : "ATF"
<5>[    0.975747] 0x0000000c0000-0x000000140000 : "Bootloader"
<5>[    0.981961] 0x000000140000-0x0000001c0000 : "Config"
<5>[    0.987806] 0x0000001c0000-0x0000002c0000 : "Factory"
<5>[    0.994260] 0x0000002c0000-0x0000006c0000 : "kernel"
<5>[    1.003601] 0x0000006c0000-0x0000075c0000 : "ubi"
```
You have flashed the wrong image, this partitioning doesn't match the supposed layout for `linksys-e8450-ubi`.

pthomas, 2022-12-04:
OK, I just did a new clone of owrt-ubi-installer, and ran the script. Is `openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb` what I want?

Can I program that with `sysupgrade` or `mtd write`?

pthomas, 2022-12-04:
OK, I was able to update using `openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb`, I had tried this before, but I don't think I was doing the `rm /sys/fs/pstore/*` each time.

Thanks for the help!

dangowrt, 2022-12-04:
> OK, I just did a new clone of owrt-ubi-installer, and ran the script. Is openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb what I want?

You do want a `*sysupgrade.itb` image for the `linksys_e8450-ubi`. You don't necessarily need to use the copy of the release image provided to you by the installer, but you can as well just use the image provided on downloads.openwrt.org or use the [OpenWrt Firmware Selector](https://firmware-selector.openwrt.org/?target=mediatek%2Fmt7622&id=linksys_e8450-ubi). Just make sure it's the right image for this device with replaced bootchain for UBI support, ie. the string `-linksys_e8450-ubi-` should be part of the filename. You can **not** use any other image on this device when using the UBI bootchain.

> Can I program that with sysupgrade or mtd write?

You should always use `sysupgrade` to update OpenWrt devices (or use the Firmware Upgrade page in LuCI Web-UI).
`mtd write` is a low-level tool and normal users should not need to interact with it directly.

> OK, I was able to update using `openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb`, I had tried this before, but I don't think I was doing the `rm /sys/fs/pstore/*` each time.

I guess what happened is that you may have flashed a wrong image (e.g. for `linksys_e8450` or `mt7622-rfb1-ubi`) and that caused the newly flashed firmware to crash (as rootfs could not be mounted). Now, with that crashlog present in pstore, the bootloader will subsequently boot to recovery mode until you clear pstore.

The weird thing for me is still that I believe that at some point you should have seen a very clear warning that should have prevented you from flashing this wrong image, ie. in LuCI there should have been a big red warnings messages and you should need place additional checkmarks there to confirm that you understood that you are about to brick the device, and when using `sysupgrade` on the command line it should not have proceeded without you giving it the `--force` option. Did you ever see such warning dialog and/or did you use the `--force` option when flashing any image in the past?

Anyway. Good to read that you managed to revive the device. If all is fine now, please close this issue.

pthomas, 2022-12-04:
Yes, I used the --force at some point :(

Thanks again for the help! This is the best wifi setup I've had at home in a long time. I'm glad to get it working again.

trashhalo, 2023-02-03:
Im in the same boat. Followed steps in the readme. router running fine until one day when I show up and its running in system recovery mode. I followed the steps on 3 routers. and now 2 are in that state.

```
root@OpenWrt:~# ubinfo -d 0 -a
ubi0
Volumes count:                           6
Logical eraseblock size:                 126976 bytes, 124.0 KiB
Total amount of logical eraseblocks:     1000 (126976000 bytes, 121.0 MiB)
Amount of available logical eraseblocks: 0 (0 bytes)
Maximum count of volumes                 128
Count of bad physical eraseblocks:       0
Count of reserved physical eraseblocks:  20
Current maximum erase counter value:     1
Minimum input/output unit size:          2048 bytes
Character device major/minor:            248:0
Present volumes:                         0, 1, 2, 3, 4, 5

Volume ID:   0 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        9 LEBs (1142784 bytes, 1.0 MiB)
State:       OK
Name:        ubootenv
Character device major/minor: 248:1
-----------------------------------
Volume ID:   1 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        9 LEBs (1142784 bytes, 1.0 MiB)
State:       OK
Name:        ubootenv2
Character device major/minor: 248:2
-----------------------------------
Volume ID:   2 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        66 LEBs (8380416 bytes, 7.9 MiB)
State:       OK
Name:        recovery
Character device major/minor: 248:3
-----------------------------------
Volume ID:   3 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        67 LEBs (8507392 bytes, 8.1 MiB)
State:       OK
Name:        boot_backup
Character device major/minor: 248:4
-----------------------------------
Volume ID:   4 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        81 LEBs (10285056 bytes, 9.8 MiB)
State:       OK
Name:        fit
Character device major/minor: 248:5
-----------------------------------
Volume ID:   5 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        744 LEBs (94470144 bytes, 90.0 MiB)
State:       OK
Name:        rootfs_data
Character device major/minor: 248:6
```

dangowrt, 2023-02-03:
Please check /sys/fs/pstore , dump what ever you find there here and delete these files. Then the router will then come up in production mode again.

irishguy90, 2023-02-04:
I'm to stuck in this but my problem is the rm /sys/fs/pstore/* says no such file or directory.

closed 2025-07-04 by dangowrt