dangowrt/owrt-ubi-installer #131: Belkin RT3200 WiFi 6

hakova, 2022-12-29:
Are these instructions not valid for the WiFi 6 RT3200? The wizard in this router would not show by direct LAN port connection unless one is connected to the router via WiFi. The wizard then refuses to proceed unless there is internet on the WAN port. Please help.

Thanks!

dangowrt, 2022-12-29:
Yes, this is for the Belkin WiFi 6 router RT3200. The new firmware from factory no longer allows to abort the wizard, you will have to at least provide DHCP to the WAN port and set the admin password before it will allow you to install OpenWrt *using the `_signed` image*.

hakova, 2022-12-29:
Thank you for your quick response. Since this was a brand new product, I assumed that it would have >= 1.2 firmware, which I couldn't see without getting rid of the wizard first. However, I bit the bullet and exit the wizard and then saw that it had firmware 1.010 on it. I was then able to follow the guide without much difficulty and set it up as a dumb AP.  I am actually connected to it right now :). Big thank you for providing this amazing software. It is mind blowing how much the consumer is restricted by commercial products.

closed 2022-12-29 by hakova