dangowrt/owrt-ubi-installer #134: RT3200 (E8450) looses settings after reboot (initial install)

master5597, 2023-01-28:
I have a RT3200 (E8450) and I originally installed OpenWrt using https://downloads.openwrt.org/releases/22.03.2/targets/mediatek/mt7622/openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb and using the linked instructions at https://github.com/dangowrt/owrt-ubi-installer.  But I notices when I lost power/rebooted all the settings were gone.  I got busy and came back to it and saw that 22.03.3 was available and I tried to install `openwrt-22.03.3-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.itb`, but the web UI gave an error. (per https://github.com/dangowrt/owrt-ubi-installer/issues/133 I didn't think I needed the ubi version for updates?)  I think something isn't mounted correctly.  I've tried using the web interface to flash the ubi versions of [22.03.1](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.0.1/openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) and [22.03.2](https://downloads.openwrt.org/releases/22.03.2/targets/mediatek/mt7622/openwrt-22.03.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) seemingly successfully, but in the web UI the Firmware version always says: `OpenWrt 22.03-SNAPSHOT r19338-ae64d0624c / LuCI openwrt-22.03 branch git-22.119.37115-66994a5`.  In reading a bunch of other posts about this issue on other devices, I ran the commands below.  Hopefully they will give some insight?  I expect i missed something obvious, as I get a lot of interruptions....

```
root@OpenWrt:~# dmesg | grep -e overlay -e jffs2 -e ubi
[    1.425273] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    1.681866] 0x000000300000-0x000008000000 : "ubi"
[    2.645910] ubi0: attaching mtd3
[    3.033869] ubi0: scanning is finished
[    3.041938] ubi0: attached mtd3 (name "ubi", size 125 MiB)
[    3.047426] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[    3.054312] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[    3.061097] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[    3.068048] ubi0: good PEBs: 1000, bad PEBs: 0, corrupted PEBs: 0
[    3.074135] ubi0: user volume: 6, internal volumes: 1, max. volumes count: 128
[    3.081352] ubi0: max/mean erase counter: 2/1, WL threshold: 4096, image sequence number: 466119738
[    3.090387] ubi0: available PEBs: 0, total reserved PEBs: 1000, PEBs reserved for bad PEB handling: 20
[    3.099750] ubi0: background thread "ubi_bgt0d" started, PID 851
[    3.106793] FIT: Selected configuration: "config-1" (OpenWrt linksys_e8450-ubi)
[    3.124083] FIT:          flat_dt sub-image 0x004db000..0x004e2578 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob) 
[    3.135489] FIT:       filesystem sub-image 0x004e3000..0x009cafff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs) 
[    3.153575]  ubiblock0_4: p1(rootfs-1)
[    3.153874] block ubiblock0_4: created from ubi0:4(fit)
root@OpenWrt:~# logread -e overlay -e jffs2 -e ubi
Wed May 11 23:35:51 2022 kern.notice kernel: [    1.681866] 0x000000300000-0x000008000000 : "ubi"
Wed May 11 23:35:51 2022 kern.notice kernel: [    2.645910] ubi0: attaching mtd3
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.033869] ubi0: scanning is finished
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.041938] ubi0: attached mtd3 (name "ubi", size 125 MiB)
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.047426] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.054312] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.061097] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.068048] ubi0: good PEBs: 1000, bad PEBs: 0, corrupted PEBs: 0
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.074135] ubi0: user volume: 6, internal volumes: 1, max. volumes count: 128
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.081352] ubi0: max/mean erase counter: 2/1, WL threshold: 4096, image sequence number: 466119738
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.090387] ubi0: available PEBs: 0, total reserved PEBs: 1000, PEBs reserved for bad PEB handling: 20
Wed May 11 23:35:51 2022 kern.notice kernel: [    3.099750] ubi0: background thread "ubi_bgt0d" started, PID 851
Wed May 11 23:35:51 2022 kern.debug kernel: [    3.106793] FIT: Selected configuration: "config-1" (OpenWrt linksys_e8450-ubi)
Wed May 11 23:35:51 2022 kern.debug kernel: [    3.124083] FIT:          flat_dt sub-image 0x004db000..0x004e2578 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob)
Wed May 11 23:35:51 2022 kern.debug kernel: [    3.135489] FIT:       filesystem sub-image 0x004e3000..0x009cafff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs)
Wed May 11 23:35:51 2022 kern.info kernel: [    3.153575]  ubiblock0_4: p1(rootfs-1)
Wed May 11 23:35:51 2022 kern.info kernel: [    3.153874] block ubiblock0_4: created from ubi0:4(fit)
root@OpenWrt:~# uname -a
Linux OpenWrt 5.10.113 #0 SMP Fri May 6 19:55:52 2022 aarch64 GNU/Linux
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 00140000 00020000 "fip"
mtd2: 00100000 00020000 "factory"
mtd3: 07d00000 00020000 "ubi"
root@OpenWrt:~# ls /dev/u*
/dev/ubi0           /dev/ubi0_3         /dev/ubiblock0_4
/dev/ubi0_0         /dev/ubi0_4         /dev/ubiblock0_4p1
/dev/ubi0_1         /dev/ubi0_5         /dev/urandom
/dev/ubi0_2         /dev/ubi_ctrl
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
tmpfs                   241.5M     15.6M    225.9M   6% /
tmpfs                   241.5M     72.0K    241.4M   0% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
root@OpenWrt:~# mount
tmpfs on / type tmpfs (rw,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
pstore on /sys/fs/pstore type pstore (rw,noatime)
root@OpenWrt:~# 
```

dangowrt, 2023-01-29:
Probably there is still an recorded error log in `/sys/fs/pstore` because you wrongly flashed the non-UBI image.
If so, clear pstore using `rm /sys/fs/pstore/*` and reboot, that should fix it and allow you to boot into production firmware again.

master5597, 2023-01-29:
OK there was a single log file (dmesg-ramoops-0) in there.  After deleting it everything seems good,  Thanks!

closed 2023-01-29 by master5597