dangowrt/owrt-ubi-installer #136: Stuck in recovery mode

irishguy90, 2023-02-04:
I recently updated the firmware on the rt320 via the gui and noticed the WiFi could work so I held in the reset button to see if that would resolve it but it's didn't. Instead it booted to recovery mode and no matter what firmware I flash with it just goes back to recovery

npmussi, 2023-02-06:
I also have the same issue. WiFi was non-functional after upgrade, so I attempted to rollback via an older image, and now I am in recovery mode with flashing having no effect.

samuel21119, 2023-02-12:
Hi,
I have upgraded my Linksys E8450 to the latest 22.03.1 release, and encountered that the device is not bringing up correctly. Are there any updates with this issue?
Thanks.

dangowrt, 2023-02-20:
You have most likely flashed a wrong image. Please see https://github.com/dangowrt/owrt-ubi-installer/issues/104

irishguy90, 2023-02-20:
> You have most likely flashed a wrong image. Please see https://github.com/dangowrt/owrt-ubi-installer/issues/104

Iv read the guide tried flashing most the firmwares and used some command but they don't work 

This what comes up with some command iv tried

root@OpenWrt:~#  ubinfo -a
UBI version:                    1
Count of UBI devices:           0
UBI control device major/minor: 10:63
root@OpenWrt:~#  ubinfo -d 0 -n 5 -a
ubinfo: error!: cannot get information about UBI volume 5 on ubi0
        error 2 (No such file or directory)
root@OpenWrt:~#  hexdump -C /dev/ubi0_4 | head -n 34
hexdump: /dev/ubi0_4: No such file or directory

the files im trying are
ede-mediatek-mt7622-MTK-AX3200-MT7531-squashfs-sysupgrade
openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer
openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer
openwrt-22.03.0-rc1-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade
[ openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

root@OpenWrt:~# hexdump -C /dev/ubi0_2 | head -n 34
hexdump: /dev/ubi0_2: No such file or directory

iv even tried FW_RT3200_1.0.01.101415_prod.img and still stuck in recovery
iv also tried
Switch device power off.
3x Switch device power on for 2 seconds, then off again.
Switch device power on, the device should now boot to the alternative partition.
and also powered off for 10 seconds after using openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb still stuck in recovery. im not sure now what else as iv been trying for few days now and still stuck in recovery

irishguy90, 2023-02-20:
rm /sys/fs/pstore/* does nothing

dangowrt, 2023-02-20:
You need to do things in the right order.
So:
Step 1: `rm /sys/fs/pstore/*`
Step 2: `sysupgrade https://downloads.openwrt.org/releases/22.03.3/targets/mediatek/mt7622/openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb`

Do **not** reboot after `rm /sys/fs/pstore`, directly proceed to sysupgrade the correct file.

damo901, 2023-02-21:
rm: can't remove '/sys/fs/pstore/*': No such file or directory

damo901, 2023-02-21:
ls /sys/fs/pstore

cd /sys/fs/pstore

vim dmesg-ramoops-0
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
- dmesg-ramoops-0 1/1 100%

damo901, 2023-02-21:
used these commands from a tread i started on openwrt forum 
[https://forum.openwrt.org/t/belkin-rt3200-stuck-in-emergency/150447/11](url)
```
root@OpenWrt:~#  ubinfo -a
UBI version:                    1
Count of UBI devices:           0
UBI control device major/minor: 10:63
root@OpenWrt:~#  ubinfo -d 0 -n 5 -a
ubinfo: error!: cannot get information about UBI volume 5 on ubi0
        error 2 (No such file or directory)
root@OpenWrt:~#  hexdump -C /dev/ubi0_4 | head -n 34
hexdump: /dev/ubi0_4: No such file or directory

the files im trying are
ede-mediatek-mt7622-MTK-AX3200-MT7531-squashfs-sysupgrade
openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer
openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer
openwrt-22.03.0-rc1-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade
[ openwrt-22.03.1-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb ``
root@OpenWrt:~# hexdump -C /dev/ubi0_2 | head -n 34
hexdump: /dev/ubi0_2: No such file or directory
```

damo901, 2023-02-21:
i think its fixed. i tried these again

Using PSTORE/ramoops

    While running the production firmware enter this command in the shell

    echo c > /proc/sysrq-trigger

    Once the router has rebooted into recovery mode, clear PSTORE to make it reboot into production mode again:

    rm /sys/fs/pstore/*

craigcurtin-dev, 2023-08-13:
Just wanted to say i too had this problem - flashed two units with the files provided in this repo and was working on configuring one with VLANs - after a save and apply - ended up in this reboot loop for no apparent reason.

I followed these direction above and that appears to have resolved the issue

While running the production firmware enter this command in the shell

echo c > /proc/sysrq-trigger

Once the router has rebooted into recovery mode, clear PSTORE to make it reboot into production mode again:

rm /sys/fs/pstore/*

Craig

tech733, 2025-01-15:
I recently had this same issue going from 22.03.5 to 22.03.7 and spent a whole night trying to figure out how to fix it. This should really be documented. 
I followed these directions to fix: 

While running the production firmware enter this command in the shell:
`echo c > /proc/sysrq-trigger`
Once the router has rebooted into recovery mode, clear PSTORE and downgrade:
```
rm /sys/fs/pstore/*
sysupgrade https://downloads.openwrt.org/releases/22.03.3/targets/mediatek/mt7622/openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
```

Do not reboot after rm /sys/fs/pstore/*, directly proceed to sysupgrade the correct file.

closed 2025-07-04 by dangowrt