dangowrt/owrt-ubi-installer #137: WiFi 5G is missing

coyan, 2023-02-18:
I don't remember when 5G disappeared or what update triggered it, probably about 2 year ago. But now a new device is on the network and want this frequency.

I saw the advices at #114 and I did this steps (the <URL> is last called "mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb" located at https://github.com/dangowrt/owrt-ubi-installer/releases/ ):

1. sysupgrade -n <url>
2. Wait to restart
3. ssh login and:

```
root@OpenWrt:/# du -h /sys/fs/pstore/
0       /sys/fs/pstore/
root@OpenWrt:/# sysupgrade -n <url>
```

Wait to restart and login on luci, gone to WiFi section and obtain the same result:

[image]

Can I do something to fix it or the 5G is broken forever?

dangowrt, 2023-02-19:
Most likely you are missing firmware files. If you were are using snapshot builds, this can happen because the firmware was previously included in `kmod-mt7915e` package and now was moved into a separate package `kmod-mt7915-firmware`. Installing `kmod-mt7915-firmware` should fix it.

coyan, 2023-02-20:
I think that the package is installed by default:
[image]

Or you suggest that I install the "_openwrt-22.03.3-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin_" file located here? https://openwrt.org/toh/linksys/e8450

Sorry but I don't know where is the installer for `kmod-mt7915-firmware` :(

dangowrt, 2023-02-20:
If you use OpenWrt 22.03 the firmware for MT7915 is part of `kmod-mt7915e` which you have installed.

> Or you suggest that I install the "_openwrt-22.03.3-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin_" file located here?

No, using a non-UBI firmware file on the device after the bootloader has been switched to the UBI-aware version **will brick your device**. Don't do it.

coyan, 2023-02-21:
I restarted the device (usin LuCi) and here are both logs:

### System Log
```Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: started, version 2.86 cachesize 150
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 8 addresses
Tue Feb 21 20:45:25 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Feb 21 20:45:25 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Tue Feb 21 20:45:25 2023 kern.info kernel: [   15.511109] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
Tue Feb 21 20:45:25 2023 kern.info kernel: [   15.518658] br-lan: port 1(lan1) entered blocking state
Tue Feb 21 20:45:25 2023 kern.info kernel: [   15.523894] br-lan: port 1(lan1) entered forwarding state
Tue Feb 21 20:45:25 2023 daemon.notice netifd: Network device 'lan1' link is up
Tue Feb 21 20:45:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 192.168.1.110 a5:47:f8:6e:55:10
Tue Feb 21 20:45:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.110 a5:47:f8:6e:55:10
Tue Feb 21 20:45:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.110 a5:47:f8:6e:55:10
Tue Feb 21 20:45:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.110 a5:47:f8:6e:55:10 computer
Tue Feb 21 20:45:32 2023 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=2427 channel=4
Tue Feb 21 20:45:32 2023 kern.info kernel: [   22.292186] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Tue Feb 21 20:45:32 2023 kern.info kernel: [   22.298674] br-lan: port 5(wlan0) entered blocking state
Tue Feb 21 20:45:32 2023 kern.info kernel: [   22.304017] br-lan: port 5(wlan0) entered forwarding state
Tue Feb 21 20:45:32 2023 daemon.notice netifd: Network device 'wlan0' link is up
Tue Feb 21 20:45:32 2023 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Tue Feb 21 20:45:32 2023 daemon.notice hostapd: wlan0: AP-ENABLED
Tue Feb 21 20:45:33 2023 daemon.info dnsmasq-dhcp[1]:    {Redacted some DHCP discoveries}
Tue Feb 21 20:46:23 2023 user.info : luci: accepted login on / for root from 192.168.1.112
```

### Kernel Log
```[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.146 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19777-2853b6d652) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Oct 7 23:34:56 2022
[    0.000000] Machine model: Linksys E8450 (UBI)
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[... cut ...]
[   14.434508] device wlan0 entered promiscuous mode
[   15.511109] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   15.518658] br-lan: port 1(lan1) entered blocking state
[   15.523894] br-lan: port 1(lan1) entered forwarding state
[   22.292186] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   22.298674] br-lan: port 5(wlan0) entered blocking state
[   22.304017] br-lan: port 5(wlan0) entered forwarding state
```

I changed the MAC for a fake one and deleted the DHCP assigments.

dangowrt, 2023-02-21:
```
[    8.830075] mtk-snand 1100d000.snfi: ECC: Uncorrectable bitflips in page 907 sect 3
[    8.837751] mt7915e 0000:01:00.0: reading EEPROM from mtd factory failed: -74
[    8.881036] mt7915e 0000:01:00.0: eeprom load fail, use default bin
[    8.887388] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_eeprom.bin failed with error -2
[    8.897043] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_eeprom.bin
[    8.912507] mt7915e: probe of 0000:01:00.0 failed with error -12
```
This is a symptom of the updated SPI-NAND driver and was discussed in https://github.com/dangowrt/owrt-ubi-installer/issues/27
Please also share your U-Boot version using `fw_printenv ver`, as that will tell the version of the installer which you were using originally. If the version is older than U-Boot 2021.10, the easiest way to fix the issue is to re-run the (now updated) installer which will take care of fixing this issue. If you are already on U-Boot 2021.10 or newer then we are dealing with a different problem and in this case please dump the EEPROM data from flash using `hexdump -C /dev/mtdblock2` and we shall analyze if it still looks good despite the supposed bit-flips and if so re-write it to fix them.

wallrik, 2023-05-23:
Anyone coming to this issue with 5G not working after an update. I just had this issue as well. However, my error seemed to be different from this issue author. My kernel log:
```
[    7.425237] mt7915e 0000:01:00.0: assign IRQ: got 146
[    7.430379] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    7.436473] mt7915e 0000:01:00.0: enabling bus mastering
[    7.558408] mtk-pcie 1a143000.pcie: msi#0 address_hi 0x0 address_lo 0x44e3d0c0
[    7.578530] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[    7.588492] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[    7.874302] mt7915e: probe of 0000:01:00.0 failed with error -12
```
I was able to fix this by installing `kmod-mt7915-firmware` from software packages and rebooting 👌

wit-know, 2024-03-06:
I too have the very same problem outlined by the OP. I have a brand new E8450 that I unwrapped yesterday and tested. Then, after reading all the text on the UBI installer, I uploaded the installer and then flashed the current 23.05.0 firmware. I'll admit that I uploaded the wrong installer initially (using the newest 1.1.1 version), but then reread everything and attempted to install the 23.05.0 firmware again. I am now confident I have the 23.05.0 firmware installed correctly.

HOWEVER, just like OP's screenshot above, I only have radio0 available, and the 5GHz radio1 is not being detected. My log looks like this:
```
root@OpenWrt:~# logread
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Mon Oct  9 21:45:37 2023 kern.notice kernel: [    0.000000] Linux version 5.15.134 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23497-6637af95aa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Oct 9 21:45:35 2023
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Machine model: Linksys E8450 (UBI)
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Zone ranges:
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   DMA32    empty
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   Normal   empty
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Early memory node ranges
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.4
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] percpu: Embedded 17 pages/cpu s30040 r8192 d31400 u69632
Mon Oct  9 21:45:37 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s30040 r8192 d31400 u69632 alloc=17*4096
Mon Oct  9 21:45:37 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 843419
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Mon Oct  9 21:45:37 2023 kern.notice kernel: [    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n1 swiotlb=512
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Memory: 502064K/524288K available (8384K kernel code, 904K rwdata, 1416K rodata, 448K init, 308K bss, 22224K reserved, 0K cma-reserved)
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000]       Tracing variant of Tasks RCU enabled.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] Root IRQ handler: 0xffffffc008423cb4
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 12.50MHz (phys).
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2e2049cda, max_idle_ns: 440795202628 ns
Mon Oct  9 21:45:37 2023 kern.info kernel: [    0.000001] sched_clock: 56 bits at 12MHz, resolution 80ns, wraps every 4398046511080ns
[... cut ...]
Mon Oct  9 21:46:42 2023 authpriv.info dropbear[2982]: Child connection from 192.168.1.254:62903
Mon Oct  9 21:46:49 2023 authpriv.notice dropbear[2982]: Auth succeeded with blank password for 'root' from 192.168.1.254:62903
root@OpenWrt:~#
```

As others have suggested in past posts, I have also checked that the kmod-mt7915-firmware is installed. I've also tried force-reinstalling it to no avail. 

Where do I go from here? I am still newish to OpenWRT, and did not expect this device to be so troublesome.

wit-know, 2024-03-06:
Also, here is the hexdump info:

```
root@OpenWrt:~# hexdump -C /dev/mtdblock2
00000000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
00000010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
00000040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00020000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
00020010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
00020020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00020030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
00020040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00040000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
00040010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
00040020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00040030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
00040040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00060000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
00060010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
00060020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00060030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
00060040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00080000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
00080010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
00080020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00080030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
00080040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000a0000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
000a0010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
000a0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000a0030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
000a0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000c0000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
000c0010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
000c0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000c0030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
000c0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000e0000  55 42 49 23 01 00 00 00  00 00 00 00 00 00 00 00  |UBI#............|
000e0010  00 00 08 00 00 00 10 00  59 a3 a9 28 00 00 00 00  |........Y..(....|
000e0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000e0030  00 00 00 00 00 00 00 00  00 00 00 00 8f 90 c1 70  |...............p|
000e0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00100000
root@OpenWrt:~#
```

wallrik, 2024-03-06:
@wit-know your issue can be seen in your kernel log:
```
[    7.005004] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220929104113a
[    7.005004]
[    7.201929] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[    7.254751] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205
[    7.371350] mt7915e 0000:01:00.0: eeprom load fail, use default bin
[    7.377846] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_eeprom.bin failed with error -2
[    7.387537] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_eeprom.bin
[    7.403893] mt7915e: probe of 0000:01:00.0 failed with error -12
```
Just searching for mt7915e on GitHub shows lots of people having issues with this driver.
I don't think this is necessarily related to this installer repository.
Here's someone with your exact error message: openwrt/openwrt#14207

dangowrt, 2024-03-06:
@wit-know What you are seeing is the (expected) result of having used the v1.1.x installer and then trying to use an older OpenWrt release. That can't work. And it's not related to the original issue, but actually another issue.

> Where do I go from here? I am still newish to OpenWRT, and did not expect this device to be so troublesome.

After using the v1.1.x installer there is no going back, or at least it is not trivial. You cannot use 23.05.x then, unless you **first completely restore the device to factory state** and then run the pre-v1.1.x installer.

If you want to do that, you will need the backup files from the `boot_backup` UBI volume and write them to the appropriate MTD partitions. If you are not sure what you are doing, probably the best for now is to just use snapshot images until there will be the next stable release.

dangowrt, 2024-03-06:
@wallrik Neither `kmod-mt7915e` nor missing firmware files are the problem here. The problem is that OP has moved the factory data into a UBI volume (instead of a raw flash partition) and is trying to run an OpenWrt which still expects it in a raw MTD partition. Simple as that.

wit-know, 2024-03-07:
> @wit-know What you are seeing is the (expected) result of having used the v1.1.x installer and then trying to use an older OpenWrt release. That can't work. And it's not related to the original [...]

Thank you, Dan, for all your help. I was able to revert to revert back to the Linksys firmware 1.0.01.101415, and then proceeded to upload the 23.05.0 installer. However, while flashing the the 23.05.0 sysupgrade, the unit bricked! I don't know how that happened, but now I have a USB UART adapter on the way and then will proceed from there.

As a side note, if I can recall correctly, after I accidentally installed the v1.1.1 installer (my fault), I was still never able to install the v1.1.1 sysupgrade successfully. That is how I ended up flashing the 23.05.0 sysupgrade instead. Regardless, I was a bit green in rushing about while installing those releases, and probably did not fully read your instructions on Github. Your detailed instructions are actually written quite clearly.

dangowrt, 2024-03-07:
@wit-know Strange, but hard to tell what happened before connecting the serial. Feel free to reach out should you need more assistance.

zow2023, 2024-06-02:
Can you tell me the details of the process of restoring the factory firmware from the latest installer？ @dangowrt

kratanat, 2024-07-19:
I also have the same issue where I ran the 1.1.2, then install the latest UBI SNAPSHOT from the OpenWRT website. I'm using Linksys e8450. After comparing configuration with Belkin 3200. I was able to find out what causing the radios to show generic. 
When I run command "iw phy" on the snapshot, and the wifi was listed under "phy0" and "phy1", while under previous stable release, iw command returned "wl0" and "wl1"

So I edited "/etc/config/wireless" 	
under
wifi-device 'radio0'
option phy 'wl0' changed to option phy 'phy0' 
under radio1 section, I change from 'wl1' to 'phy1'

After refresh wifi page on luci, I am able to see the correct name 
[image]

However, it seems like there is an automatic generation of wireless config that it added a wl0 and wl1 entries back. but it did work.

[image]

john-clark, 2024-11-27:
What is the solution for the generic 5ghz wifi not loading wth the 23.05.05 and installer 1.1.1?

My router also will not keep its dhcp lan and requests a new ip every reboot with the same mac address.

VelvetBcuz, 2024-12-08:
Until and unless this problem is solved, what is the best stable version available today?

dangowrt, 2025-07-04:
OpenWrt 24.10.2 after running installer v1.1.3 or v1.1.4 are known to work well.

closed 2025-07-04 by dangowrt