dangowrt/owrt-ubi-installer #139: Linksys v1.1.01.272918 can't downgrade

Hadyark, 2023-03-24:
I do know why i can't downgrade to the [FW_E8450_1.0.01.101415_prod.img] (https://downloads.linksys.com/support/assets/firmware/FW_E8450_1.0.01.101415_prod.img) ? I'm currently in v1.1.01.272918

[image]

dangowrt, 2023-03-24:
Strange, I always thought that signature checking is only active in version 1.0.02.*, but it looks like signature check is now enabled also in this firmware you got there already. If you just want to install OpenWrt, use the `*_installer_signed.itb`, it should work.

Hadyark, 2023-03-25:
Thanks, it worked for me

dahu33, 2023-03-30:
Same here with firmware version 1.1.01 build 272918 Mar 29, 2022 (device bought in Japan). Using `openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb` worked.

ptrlow1, 2025-03-04:
On a E8450 it's using 1.1.01.272198, (device I bought was refurbished through woot! in the US) the unsigned image file won't work. However if `openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb` is uploaded, it works again.

closed 2025-07-04 by dangowrt