dangowrt/owrt-ubi-installer #143: E7350 Support

LokeYourLord, 2023-04-07:
As the title states. The E7350 is a slightly weaker version of the E8450, and as such I think supporting it should be fairly straight forward.

Also: If anyone can, please post a picture of the UART Serial layout on your E8450 device that I can use as reference for my E7350, thank you.

dangowrt, 2023-04-07:
Unfortunately the Linksys E7350 (aka. Belkin RT1800) is a quite different device. It is using the MT7621 dual-core MIPS SoC rather than MT7622 dual-core ARM Cortex-A53.
The MT7621 supports booting off SPI-NAND only with [some special tricks available only on some SPI-NAND chips and require an intermediate preloader to then load U-Boot](https://github.com/hackpascal/mt7621-boot-spi-nand). This is very different from MT7622 which has native support for SPI-NAND and being an ARMv8 CPU it is using ARM TrustedFirmware-A, in this case also as a preloader which can easily be built form source.

I'm not saying it's impossible to support the E7350 (aka. RT1800), but it would require attaching JTAG to have a recovery method while developing a bootchain based on the SPI-NAND preloader and U-Boot...

dangowrt, 2023-04-07:
Serial pinout see here:
https://openwrt.org/toh/linksys/e8450#serial

LokeYourLord, 2023-04-07:
I see. Well the pinout is also totally different as I see, and I for some reason cannot get into UART Serial U-Boot on my device. Maybe the pins I use are incorrect (wrong order or setup), or something else.

Is there a specific way to make serial output show up? Like, do I first need to push a button after powering on, or what else is there for me to do to enter serial? Maybe I can help put up a guide with my device if I can find out how to enter the bootloader first.

gretthunberg, 2023-04-21:
I have a router with the same SoC but with a non-Micron NAND chip, so that option is out for me.  Does this project support booting off parallel NAND?

LokeYourLord, 2023-04-22:
> I have a router with the same SoC but with a non-Micron NAND chip, so that option is out for me.  Does this project support booting off parallel NAND?

I flashed OpenWRT via UART Serial, and I'd recommend you do the same. As of now, you cannot use this project to flash the E7350. I've written an extensive flashing guide on the OpenWRT forums, here: https://forum.openwrt.org/t/linksys-e7350-openwrt-install/156673/64?u=lokeyourc3ph

closed 2024-01-14 by LokeYourC3PH