dangowrt/owrt-ubi-installer #150: Connection establishment takes long time

txfs19260817, 2023-08-25:
Hi there, thanks for the great work!

I just installed and upgraded openwrt on my new hardware. However, I found connect to a new website can take long time even timed out; for some apps like ChatGPT it can timeout if I try to resume a dialogue after opening for a while (but Youtube stream is fine). This problem applies to all my connected devices.

At first I thought it's due to DNS lookup, then I disabled DNS peer advertising and added DNS nameservers like 1.1.1.1, but it didn't solve this problem. 

I also observed from Chrome that [Initial connection](https://developer.chrome.com/docs/devtools/network/reference/?utm_source=devtools#timing-explanation) takes around 16 seconds whenever connecting to a new website.

I really appreciate it if you can share some insights on how to diagnose this problem!

---
```
Model | Linksys E8450 (UBI)
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/mt7622
Firmware Version | OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.119.80898-65ef406
Kernel Version | 5.10.176
```

dangowrt, 2023-08-25:
I would guess your issue is related to IPv6 not being routed. The reason could be that your ISP does offer IPv6 connectivity, but does not support prefix-delegation and only announces a single /64 prefix.
For the devices behind the OpenWrt router to have IPv6 connectivity you will need to setup NDP proxy, selecting the WAN as upstream interface and choosing "relay" both all IPv6 DHCP modes on both, LAN and WAN interface.
See https://openwrt.org/docs/guide-user/network/ipv6/configuration#ipv6_relay for an example.

Anyway, first make sure that really missing IPv6 is the problem.

txfs19260817, 2023-08-26:
Thank you for your reply! However, it didn't seem to resolve the problem, if my setting has no wrong.

```
> cat /etc/config/dhcp
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config dhcp 'wan6'
        option interface 'wan6'
        option master '1'
        option ignore '1'
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'
```

```
> ifconfig
br-lan    Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fd34:6972:f4ac::1/60 Scope:Global
          inet6 addr: fe80::ea9f:80ff:fee9:5f82/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:19915 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19485 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4115071 (3.9 MiB)  TX bytes:17908617 (17.0 MiB)

eth0      Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          inet6 addr: fe80::ea9f:80ff:fee9:5f82/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:46287 errors:0 dropped:0 overruns:0 frame:1
          TX packets:44483 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:22725229 (21.6 MiB)  TX bytes:22171717 (21.1 MiB)
          Interrupt:37

lan1      Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:20532 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19517 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4196791 (4.0 MiB)  TX bytes:17911718 (17.0 MiB)

lan2      Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan3      Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan4      Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:82
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:93 errors:0 dropped:0 overruns:0 frame:0
          TX packets:93 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9189 (8.9 KiB)  TX bytes:9189 (8.9 KiB)

wan       Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:81
          inet addr:192.168.1.194  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2601:647:4b01:242f::c2/128 Scope:Global
          inet6 addr: fe80::ea9f:80ff:fee9:5f81/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:25751 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18299 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:17694894 (16.8 MiB)  TX bytes:3477709 (3.3 MiB)

wlan0     Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:83
          inet6 addr: fe80::ea9f:80ff:fee9:5f83/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:393 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:53244 (51.9 KiB)

wlan1     Link encap:Ethernet  HWaddr E8:9F:80:E9:5F:84
          inet6 addr: fe80::ea9f:80ff:fee9:5f84/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:32 errors:0 dropped:0 overruns:0 frame:0
          TX packets:379 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2783 (2.7 KiB)  TX bytes:52117 (50.8 KiB)
```

dangowrt, 2023-08-26:
Well, it looks like Comcast is only handing out a single address to the device.

Try `option reqprefix '64'` in the network interface, sometimes it can help to make it explicit that you want a prefix and 64-bit would be ok (`2601:647:4b01:242f::/64` is most likely assigned to you right now, and `0xc2` == `194`, so that's a DHCPv6 server which assigned idential local parts for IPv4 and IPv6. So in theory, with DHCPv6 relay it should work.

Can you ping IPv6 hosts from clients behind the router?

txfs19260817, 2023-08-26:
Hi, I tried to add `option reqprefix '64'` under wan6 in `/etc/config/network`, unfortunately it still doesn't work.

`ping -6 google.com` works on router but timed out on my laptop. It turns out that no IPv6 address assigned to my devices  when joined router's network.

Then I joined back to my old modem-router provided by ISP and this time my devices do have a IPV6 address. Yet `ping -6 google.com` returns `Destination net unreachable`.

dangowrt, 2023-08-28:
Ok, so IPv6 connectivity is impaired somehow. The best you can do to at least have working IPv4 and no longer experience long delays caused by IPv6 connection attempts is to disabled the `wan6` interface.