dangowrt/owrt-ubi-installer #151: RT new firmware ver.: Ver. 1.2.00.360516

theowlpo, 2023-09-17:
Hi Team,

I bought a new RT3200 which pre-loaded firmware version is 1.2.00.360516.
Following all steps described, either using the signed or unsigned, the router does not allow to upload and flash the openwrt image (downloaded from this github repo).

Is it possible to update the openwrt image so will be possible to flash the RT3200?

Thank you very much!

mrkiko, 2023-09-17:
> Hi Team,
>
> I bought a new RT3200 which pre-loaded firmware version is 1.2.00.360516.
> Following all steps described, either using the signed or unsigned, the router does not allow to upload and flash [...]

Hi!

may I ask you what's the image name you tried to upload (both signed and unsigned)?
And what's exactly the error message you get back?

thanks,
Enrico

theowlpo, 2023-09-17:
Hi Enrico,

This was the signed image i have tried to flash:
openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb

unsigned:
openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb

The router replied: invalid image

Both images were downloaded from here: https://github.com/dangowrt/owrt-ubi-installer/releases

Reading the vendors note from the download section, it says that downgrading from current version is not possible. Don't know if this affect openwrt images... website here: https://www.belkin.com/support-article/?articleNum=208567

Thank you!

dangowrt, 2023-09-17:
Ok, so surprisingly things were working just fine with 1.2.00.273012 but now with 1.2.00.360516 not any more. It looks like the new firmware still uses the same GPG encryption key (which is still contained in the firmware) and even the `sbin/fwupgrade` executable is still identical. To make sure we are eliminating all other possible causes for problems please make sure:
* use Firefox Desktop browser for the whole procedure
* do a factory reset of the router
* make sure router is **not** connected to the Internet
* now upload `openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb`

I might try myself on my hardware a bit later.

theowlpo, 2023-09-17:
> Ok, so surprisingly things were working just fine with 1.2.00.273012 but now with 1.2.00.360516 not any more. It looks like the new firmware still uses the same GPG encryption key (which is still [...]

Hi dangowrt,

I have followed your recomended steps above but no success. Router stills show same message: "The image file is         incorrect".

I have tried with both openwrt imagex, signed and unsigned.

Thank you.

theowlpo, 2023-09-18:
> Ok, so surprisingly things were working just fine with 1.2.00.273012 but now with 1.2.00.360516 not any more. It looks like the new firmware still uses the same GPG encryption key (which is still [...]

Hi Daniel,

Any news on this subject? Did you have a go in your HW?
Just to make sure that is not my mistake on the procedure to flash the RT3200...

Thank you for your support.

dangowrt, 2023-09-18:
> Any news on this subject? Did you have a go in your HW?

I've tried now all the way: Restored factory flash content with Firmware 1.0.00, then updated to 1.1.01.272918 which does the switch to the new "signed" firmware format, then update to 1.2.00.360516.
From there I uploaded `openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb` and it worked flawlessly, OpenWrt's recovery system is greeting me seconds later.

However, for the firmware to be able to flash *anything* (even original vendor binaries) I had to do a factory reset while running 1.2.00.360516 and connect the router to the Internet at least once (probably for NTP to set date and time for signature verification to work...).

theowlpo, 2023-09-18:
> Restored factory flash content with Firmware 1.0.00

Thanks for the description. Just one question. How do you restore factory flash with firmware 1.0.00?

dangowrt, 2023-09-18:
> How do you restore factory flash with firmware 1.0.00?

I restore the complete content of the flash from an earlier backup. To make it easier I got the serial console connected while doing that, also so that I can see what goes wrong in case something does go wrong.

theowlpo, 2023-09-19:
Thank you. I'll try to do it tonight.

theowlpo, 2023-09-19:
Hi connected the serial, did a factory reset and can't get in the terminal the menu to flash from TFTP....

Thanks for the help

Here is the dump:
F0: 102B 0000

F6: 0000 0000

V0: 0000 0000 [0001]

00: 0000 0000

BP: 0000 0041 [0000]

G0: 0190 0000

T0: 0000 036D [000F]

Jump to BL

UNIVPLL_CON0 = 0xFE000000!!!

mt_pll_init: Set pll frequency for 25M crystal

RAM_CONSOLE preloader last status: 0x0 0x0 0x0 0x0 0x0 0x0 

[PMIC_WRAP]wrap_init pass,the return value=0.

[pmic_init] Preloader Start..................

[pmic_init] MT6380 CHIP Code, reg_val = 0, 1:E2  0:E3

[pmic_init] Done...................

Chip part number:7622B

MT7622 Version: 1.2.8, (iPA) 

SSC OFF

mt_pll_post_init: mt_get_cpu_freq = 1350000Khz

mt_pll_post_init: mt_get_mem_freq = 1600000Khz

mt_pll_post_init: mt_get_bus_freq = 1119920Khz

[PLFM] Init I2C: OK(0)

[BLDR] Build Time: 20200522-165358

==== Dump RGU Reg ========

RGU MODE:     14

RGU LENGTH:   FFE0

RGU STA:      40000000

RGU INTERVAL: FFF

RGU SWSYSRST: 8000

==== Dump RGU Reg End ====

RGU: g_rgu_satus:2

 mtk_wdt_mode_config  mode value=10, tmp:22000010

PL RGU RST: ??
[... 525 lines cut ...]

[    6.894334]  GE 1e_174(0xb5b5), 1e_175(0xb7b5)  

[    6.947464]  GE Tx offset AnaCal Done! (pair-0)(4)(0x23) 0x1e_172=0x2320
[    6.980647]  GE Tx offset AnaCal Done! (pair-1)(3)(0x3) 0x1e_172=0x2303
[    7.013738]  GE Tx offset AnaCal Done! (pair-2)(3)(0x22) 0x1e_173=0x2220
[    7.041598]  GE Tx offset AnaCal Done! (pair-3)(2)(0x21) 0x1e_173=0x2221
[    7.120899]  GE Tx amp AnaCal Done! (pair-0)(1e_12 = 0x5816)
[    7.210366]  GE Tx amp AnaCal Done! (pair-1)(1e_17 = 0xe0e)
[    7.277374]  GE Tx amp AnaCal Done! (pair-2)(1e_19 = 0x121a)
[    7.364262]  GE Tx amp AnaCal Done! (pair-3)(1e_21 = 0xf17)
[    7.417674] PORT 4 RX_DC_OFFSET
[    7.431930] before pairA output = f
[    7.437483] after pairA output = ff
[    7.442956] before pairB output = e
[    7.448511] after pairB output = 0
[    7.453885] before pairC output = e
[    7.459433] after pairC output = fe
[    7.464901] before pairD output = f
[    7.470446] after pairD output = ff
[    7.476599] [SPM] pwrap node has not register regmap
[    7.482010] regulator regulator.3: info->modeset_reg 0xc = 0x181c32a2
[    7.489441] [SVS] DCVALUES is minus, set 0
[    7.493743] regulator regulator.3: info->modeset_reg 0xc = 0x101c32a2
[    7.507217] Waiting 1 sec before mounting root device...

theowlpo, 2023-09-19:
Hi
Success!!!
- Option 2 from menu (serial)
- Downgraded the RT3200 firmware to FW_RT3200_1.0.01.101415_prod.img (via TFTP)
- From router 192.168.1.1 updated to openwrt using: openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb (the signed file did not work...)
- Reboot
- From router 192.168.1.1 updated to openwrt using: openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
- Reboot
- Done!

Thank you for the support!!!

closed 2025-07-04 by dangowrt