dangowrt/owrt-ubi-installer #158: OpenWRT UBI Installer for more routers

agha57, 2023-10-21:
Is it possible to do similar act to get rid of ‘dual firmware’ which is, in my opinion, cumbersome and wasteful of flash RAM capacity in other Linksys routers of E or EA series, specifically “Linksys E4200 v2”?

howl, 2024-02-19:
With dual firmware I suppose you mean like in the Linksys WRT3200ACM you can boot between the previous flashed firmware and the actual. It's having half the space of the flash memory but it also provides a convenient way to not brick the device.

mrkiko, 2024-05-04:
You might implement these features by hacking your way trough the u-boot environment, even tough I admit serial access might be useful here, or using netconsole which is supported by OpenWrt u-boot.
The installer was built to make it easier to install OpenWrt on this device, and even tough it might support more devices one day, it is not meant to implement this kind of features.
After running the installer you have an environment which makes it much easier to build what you want, but it's up to you then, maybe in a different package.
Remember that - if you mess up with the u-boot environment, you can restore to fresh one.

closed 2025-07-04 by dangowrt