dangowrt/owrt-ubi-installer #162: Attended sysupgrade doesn't upgrade to latest OpenWrt version

cortadocodes, 2023-11-04:
I followed the instructions [here](https://github.com/dangowrt/owrt-ubi-installer#upgrading-to-the-latest-openwrt-release) to upgrade to the latest OpenWrt version but was only upgraded by a couple of patch versions to `22.03.5`. I can see though there's a [newer version](https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v1.0.3) in your releases with the latest OpenWrt version of `23.05.0`. Should I be upgrading a different way or do the docs need updating? Or maybe `23.05.0` just isn't available on the update server yet?

cortadocodes, 2023-11-04:
Update: I found you can choose `23.05.0` by enabling the advanced option but trying to update to this version causes these errors:
```
Server response: Error: Impossible package selection
```
```
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

lectrical, 2023-11-05:
You just install this from the LuCi web interface Backup/Flash firmware.

https://downloads.openwrt.org/releases/23.05.0/targets/mediatek/mt7622/openwrt-23.05.0-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

It is the same file as this one

https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.0.3/openwrt-23.05.0-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

I followed guide for release [v1.0.2](https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v1.0.2) then installed that image from web ui and it's been running fine on 23.05.0 ever since. You approach seems overly complicated.

cortadocodes, 2023-11-05:
Thank you I'll try that. Could you update the readme so its instructions for upgrading reflect the recommended approach?

lectrical, 2023-11-05:
I've not used the `luci-app-attendedsysupgrade` so perhaps it's there for a reason. I think i'll add it and see how it works so i understand it better though I can't replicate your issue as i'm already on 23.05.0.

https://fabianlee.org/2023/01/09/openwrt-more-convenient-sysupgrade-using-attended-sysupgrade/

The method above via the webui should also preserve (most) settings anyway and you can backup settings/list but i'm not 100% sure. If you need to preserve certain things maybe hold off until you know for sure?

cortadocodes, 2023-11-06:
That would be useful thanks! I'll read a bit more into it and then see if I can get attended upgrade working

lectrical, 2023-11-06:
So TIL `auc` is the command line tool and `luci-app-attendedsysupgrade` and `auc` kind behaves like the web tool with advanced on by default.

Have you tried the command line version?

```bash
root@rt3200:~# auc --help
auc/0.3.2-1
auc: Attended sysUpgrade CLI client
Usage: auc [-b <branch>] [-B <ver>] [-c] [-f] [-h] [-r] [-y]
 -b <branch>    use specific release branch
 -B <ver>       use specific release version
 -c             only check if system is up-to-date
 -f             use force
 -h             output help
 -n             dry-run (don't download or upgrade)
 -r             check only for release upgrades
 -F <fstype>    override filesystem type
 -y             don't wait for user confirmation

Please report issues to improve the server:
```

huyz, 2023-11-20:
I'm running into the exact same issues.

Running the command line tool `auc -b 23.05.2 -y` did in fact upgrade, unlike Attended Sysupgrade in the LuCI web.

However, for me, right off the bat, my 5GHz radio isn't coming. Not knowing how to fix this off the top of my head, I'll downgrade back to 22.03.5

tomazin, 2023-11-20:
you need run `auc -b 23.05 -B 23.05.2`
two days ago I have upgraded my device, and have same problem on 5Ghz radio. I don't know why but I needed to install  kmod-mt7915e

huyz, 2023-11-21:
> you need run auc -b 23.05 -B 23.05.2

That made no difference in my case.

> two days ago I have upgraded my device, and have same problem on 5Ghz radio. I don't know why but I needed to install kmod-mt7915e

Hmm, for me, that module is already included (as I can judge from the output of `auc…`)

----

What worked finally was to skip ASU and `auc` and just to do what [lectrical said](https://github.com/dangowrt/owrt-ubi-installer/issues/162#issuecomment-1793844893):

1. go to to the Firmware Selector at https://firmware-selector.openwrt.org/?version=23.05.2&target=mediatek%2Fmt7622&id=linksys_e8450-ubi
2. download https://downloads.openwrt.org/releases/23.05.2/targets/mediatek/mt7622/openwrt-23.05.2-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
3. In LuCI, go to `System` > `Backup / Flash Firmware` > `Flash new firmware image` and upload that file

OatmealMu, 2023-11-22:
> That made no difference in my case.
>
>
> Hmm, for me, that module is already included (as I can judge from the output of `auc…`)
>
> What worked finally was to skip ASU and `auc` and just to do what [...]

This worked for me, too. Was finally able to up the channel width from 40 to 80 MHz without 5 GHz dying on me.

cortadocodes, 2023-11-22:
> In LuCI, go to System > Backup / Flash Firmware > Flash new firmware image and upload that file

When you did this, were your settings preserved? If not, do you know if we're able to restore backups after a firmware upgrade by this method without reverting the firmware version?

huyz, 2023-11-22:
Yes my settings were preserved

cortadocodes, 2023-12-22:
**SOLUTION**: Upgrading to `23.05.x` works if you remove `luci-ssl` from the packages included in the upgrade/build.

closed 2023-12-22 by cortadocodes