dangowrt/owrt-ubi-installer #166: no lights when powering on,am i bricked

ned42, 2023-12-27:
got openwrt worked on my e8450 with ubi installer sucessfully for like a month
just cant boot on now when i powering it on and i cant see any light turning on either front or back
tried press reset button with hold and turn on for like 3-4 times but still got nothing
tired connect with a pc on 192.168.1.x to confirm whether the system is working got nothing
wanna know how to solve this kind of situation
thanks

ned42, 2024-01-06:
so i checked ttl output,it says "BL2: Failed to load image id 3" just like this:https://forum.openwrt.org/t/need-help-with-a-bricked-linksys-e8450/144836
it sounds like i need to plug 20 pins into jtag to do the recover, but in openocd-scripts [here](https://github.com/mtk-openwrt/openocd-scripts/tree/main/mt7622) says:
> If the bootloader is broken, make sure BootROM has output "System halt!" before moving to step 2

so...any chance i can get rid of 20 pins jtag? @dangowrt

usrlocalben, 2024-02-18:
There is an outline for recovering the device via JTAG here:
https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/4010

My impression is that @dangowrt is working on some new boot tech. (Also described in that thread) If progress is made there then I'll recover mine and put it in service--otherwise it goes in the trash since it's not reliable.

ned42, 2024-02-22:
@usrlocalben thanks for the tips! i've sent it to a local workshop and got it repaired by flashing some copy files(sort of programmer reflashing idk) from a new machine.
and yes, i still dont even know how it just went out during normal working, just hope this time i wont need to send it back that soon ：）

schuettecarsten, 2024-03-08:
I have the same problem - no lights turn on, but the USB port has power. For me, turning the device off, wait some seconds or minutes and then turn it on sometimes helps. If not, try again and again. So, the device is not really bricked, but the bootloader seems to be buggy.

dangowrt, 2024-03-08:
> Is that `bad cell count` output normal?

Yes, that's not related, just a formality related to DTS, no functional impact what-so-ever.

Sounds more like you are also affected by https://github.com/mtk-openwrt/arm-trusted-firmware/pull/7

dalutou, 2024-03-09:
@dangowrt

dalutou, 2024-03-09:
I have also encountered this problem because of a sudden power outage, and the E8450 has never bootup again.
After recovery with jtag, I still use the same openwrt version (23.05.0). When executing the `reboot` command, there is a high probability that it will not be able to bootup.

But after adding the following command to `/etc/rc.local`

```
echo 812500 > /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
```

The problem never appeared again.

Remove `scaling_min_freq` setting, bootup failure will occur again.

@dangowrt

schuettecarsten, 2024-03-09:
> But after adding the following command to `/etc/rc.local`
> 
> ```
> echo 812500 > /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
> ```

Do you also set a different governor?

I also found this:
https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/4062

@dangowrt

schuettecarsten, 2024-03-09:
My device is in a very strange state now. I can set whatever I want, change CPU frequencies or governors, nothing seems to help yet. On every reboot, regardless if it is a soft or a hard reboot, the device seems to be dead. Solution is to unplug it competely, wait 5 to 10 minutes, reconnect and it will boot.

dalutou, 2024-03-09:
> My device is in a very strange state now. I can set whatever I want, change CPU frequencies or governors, nothing seems to help yet. On every reboot, regardless if it is a soft or a hard reboot, the device seems to be dead. Solution is to unplug it competely, wait 5 to 10 minutes, reconnect and it will boot. On the next reboot, it will look dead again, so unplug, wait, and it boots again....

* I have not set a different governor, just keep it default.
* My second E8450 bricked this morning, can not bootup, got the error `ERROR:   BL2: Failed to load image id 5 (-2)`
* I fix it with `uartboot`.
* Maybe it's due to an issue with uboot, so I upgraded it to owrt v1.1.1.  At least everything is ok now.

schuettecarsten, 2024-03-09:
Is it possible to update the bootloader from a running OpenWRT system?
I do not want to run into reboot issues while upgrading using the installer and my device is not yet "opened".

dangowrt, 2024-03-09:
> Is it possible to update the bootloader from a running OpenWRT system?

Yes, by force-flashing the v1.1.0-or-later installer.

schuettecarsten, 2024-03-09:
> > Is it possible to update the bootloader from a running OpenWRT system?
> 
> Yes, by force-flashing the v1.1.0-or-later installer.

But then I have to reboot - i hoped it is possible to flash a possibly fixed bootloader directly without needing to boot from a broken one.

dangowrt, 2024-03-09:
I get your point, but it's not possible because what the installer does is much more than just writing a new bootloader. The size of the UBI MTD partition also changed, and `factory` and `fip` need to be written into static UBI volumes instead of MTD partitions...

schuettecarsten, 2024-03-09:
Ok, and if I already ran the 1.1.0 installer some weeks ago, as I am using custom snapshot builds?

dangowrt, 2024-03-09:
> Is manual update to the 1.1.1 bootloader possible?

Yes, that's easy, but not needed. You can do it in the bootloader menu while providing the files via TFTP.

schuettecarsten, 2024-03-09:
Hmm, ok. So I hope for a good idea and a fix. 🤷