dangowrt/owrt-ubi-installer #174: Rookie installs debugging build by accident and cannot reverse it.

rshambar, 2024-03-07:
Hello, in an effort to flash my Belkin RT3200 (aka Linksys e8450) I accidentally use the 'v1.1.1: also build bl2 image for debugging issue', I flashed the '...recovery-installer.itb', then the '...squashfs-sysupgrade.itb'. I now cannot get to a UI and while I can SSH into the router I cannot seem to download or install an update to overwrite what I have flashed.  Is there a path to re-flash this with the latest v1.0.3.  I am very novice with the command line so any direction and support you could provide would be terrific, thanks!

samuel-w, 2024-03-09:
Since you've flashed the 1.1.1 version, you can only use snapshot builds of OpenWRT. Go to https://firmware-selector.openwrt.org, select E8450 UBI and SNAPSHOT, click "Customize installed packages and/or first boot script" (add any packages you want here), and click REQUEST BUILD. This will create an image that includes the Luci web UI.

Copy the firmware to your router with scp (something like ```scp <name and location of file> root@192.168.1.1:/tmp/ from your computer) and flash it with ```sysupgrade /tmp/<name of file copied>```.

Alternatively, you can run ```opkg update``` and ```opkg install luci``` through SSH to get the UI, but it might not work.

rshambar, 2024-03-10:
Thanks for the help, you are right it wouldn't let me downgrade and the opkg failed to download.

ShiromMakkad, 2024-03-25:
@samuel-w I'm in the same boat. I successfully followed your instructions, but I'm not seeing the webUI at 192.168.1.1. I did get internet restored to my PC though! I don't know what the SSID or password to get my other devices working for now. 

Unlike rshambar, I didn't flash '...squashfs-sysupgrade.itb' before losing internet connection during the first flashing process. Not sure if that matters since your instructions seem like it's doing the same thing over ssh. If it helps, here are the packages that were in the build:

```
base-files busybox ca-bundle dnsmasq dropbear firewall4 fitblk fstools kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7622-firmware kmod-mt7915-firmware kmod-nft-offload kmod-usb3 libc libgcc libustream-mbedtls logd luci mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls
```

closed 2025-07-04 by dangowrt