dangowrt/owrt-ubi-installer #177: Stuck on the same production image after commit 6aec3c7

amaumene, 2024-03-19:
Hi,

followed your [procedure](https://github.com/openwrt/openwrt/commit/6aec3c7b5bf5e5a999a12121dfa71963afb6f003#commitcomment-139151946) and manage to upgrade to r25357-ef34f8f5b5. But it seems I'm now stuck on this version. I sysupgraded to r25396-3e6f64d443 successfully but after reboot, I'm still running r25357-ef34f8f5b5.

This is dmesg on recovery image:
`[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n1 swiotlb=512 ubi.block=0,fit root=/dev/fit0
[    1.220168] 0x000000080000-0x000008000000 : "ubi"
[    1.358285] ubi0: default fastmap pool size: 50
[    1.362849] ubi0: default fastmap WL pool size: 25
[    1.367639] ubi0: attaching mtd1
[    1.647349] ubi0: scanning is finished
[    1.655886] ubi0: attached mtd1 (name "ubi", size 127 MiB)
[    1.661408] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[    1.668276] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[    1.675061] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[    1.682022] ubi0: good PEBs: 1020, bad PEBs: 0, corrupted PEBs: 0
[    1.688107] ubi0: user volume: 8, internal volumes: 1, max. volumes count: 128
[    1.695324] ubi0: max/mean erase counter: 54/34, WL threshold: 4096, image sequence number: 1416148056
[    1.704625] ubi0: available PEBs: 0, total reserved PEBs: 1020, PEBs reserved for bad PEB handling: 20
[    1.713933] ubi0: background thread "ubi_bgt0d" started, PID 250
[    1.733987] block ubiblock0_5: created from ubi0:5(fit)
[    3.064433] FIT: Selected configuration: "config-1" (OpenWrt linksys_e8450-ubi)
[    3.081509] FIT:          flat_dt sub-image 0x0057b000..0x00582ab3 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob)
[    3.092906] FIT:       filesystem sub-image 0x00583000..0x0149ffff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs)
[    3.104156] block ubiblock0_5: mapped 1 uImage.FIT filesystem sub-image as /dev/fit0
[    3.304530] ubi: mtd1 is already attached to ubi0`

and on production:
`[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n1 swiotlb=512 ubi.block=0,fit root=/dev/fit0
[    0.617530] 0x000000080000-0x000008000000 : "ubi"
[    0.755370] ubi0: default fastmap pool size: 50
[    0.759945] ubi0: default fastmap WL pool size: 25
[    0.764738] ubi0: attaching mtd1
[    1.044773] ubi0: scanning is finished
[    1.053229] ubi0: attached mtd1 (name "ubi", size 127 MiB)
[    1.058731] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[    1.065617] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[    1.072409] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[    1.079363] ubi0: good PEBs: 1020, bad PEBs: 0, corrupted PEBs: 0
[    1.085452] ubi0: user volume: 8, internal volumes: 1, max. volumes count: 128
[    1.092669] ubi0: max/mean erase counter: 54/34, WL threshold: 4096, image sequence number: 1416148056
[    1.101970] ubi0: available PEBs: 0, total reserved PEBs: 1020, PEBs reserved for bad PEB handling: 20
[    1.111279] ubi0: background thread "ubi_bgt0d" started, PID 254
[    1.130699] block ubiblock0_5: created from ubi0:5(fit)
[    2.453450] FIT: Selected configuration: "config-1" (OpenWrt linksys_e8450-ubi)
[    2.470526] FIT:          flat_dt sub-image 0x0057b000..0x00582ab3 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob)
[    2.481923] FIT:       filesystem sub-image 0x00583000..0x0149ffff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs)
[    2.493169] block ubiblock0_5: mapped 1 uImage.FIT filesystem sub-image as /dev/fit0
[    2.669866] ubi: mtd1 is already attached to ubi0
[    6.269635] UBIFS (ubi0:7): Mounting in unauthenticated mode
[    6.275407] UBIFS (ubi0:7): background thread "ubifs_bgt0_7" started, PID 626
[    6.299508] UBIFS (ubi0:7): recovery needed
[    6.433083] UBIFS (ubi0:7): recovery completed
[    6.437616] UBIFS (ubi0:7): UBIFS: mounted UBI device 0, volume 7, name "rootfs_data"
[    6.445468] UBIFS (ubi0:7): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[    6.455398] UBIFS (ubi0:7): FS size: 85454848 bytes (81 MiB, 673 LEBs), max 684 LEBs, journal size 4317184 bytes (4 MiB, 34 LEBs)
[    6.467073] UBIFS (ubi0:7): reserved for root: 4036243 bytes (3941 KiB)
[    6.473694] UBIFS (ubi0:7): media format: w5/r0 (latest is w5/r0), UUID E12BEEB1-B68D-4EC0-8423-B92228062BE0, small LPT model
[    6.486762] mount_root: switching to ubifs overlay`

Any ideas on how to solve this are welcome :)

dangowrt, 2024-03-19:
You are missing the `fitblk` package which is now needed to be able to perform `sysupgrade` with the new layout.
This is (unfortunately) common in case you were generating the new image using `auc` or `luci-app-attendedsysupgrade` based on a version which was still using the (now deprecated) uImage.FIT partition parser instead of the new `fitblk` driver which replaces it.

The fix is easy:
```
opkg update
opkg install fitblk
```

amaumene, 2024-03-19:
Nevermind I forgot to add fitblk to my custom image. Closing this issue.

closed 2024-03-19 by amaumene

amaumene, 2024-03-19:
> You are missing the `fitblk` package which is now needed to be able to perform `sysupgrade` with the new layout. This is (unfortunately) common in case you were generating the new image using `auc` [...]

Sorry I didn't see your comment. Thanks for your super quick answer though!