dangowrt/owrt-ubi-installer #179: Belkin RT3200 Stuck in Recovery Mode can only force upgrade (which all fail)

YourTheFinestIveEverKnown, 2024-03-27:
Initially I ran the 23.05.0 and got nothing after the reboot. I thought it was because I uploaded the unsigned 23.05.0 build.
Did a reset and the recovery mode comes back. Tried the latest signed build but was greeted by:

[ Wed Mar 27 02:17:10 UTC 2024 upgrade: Image metadata not present Wed Mar 27 02:17:10 UTC 2024 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware Invalid image type. Image check failed.

The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform.

Image check failed:

Wed Mar 27 02:17:10 UTC 2024 upgrade: Image metadata not present Wed Mar 27 02:17:10 UTC 2024 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware Invalid image type. Image check failed.

 Force upgrade

Select 'Force upgrade' to flash the image even if the image format check fails. Use only if you are sure that the firmware is correct and meant for your device!](url)

-The time was default (1970...etc) at first, don't know if that even matters. So I thought it could be that, but after a time sync it's still not liking the signed firmwares and I'm still having the same problem. Forcing an upgrade, router doesn't respond, I do a factory reset and get back into WRT Recovery to try again.

-So far I tried 23.05.0 signed, 22.03.3 signed, and 22.03.1 signed, but getting the same message shown above. I tried to force the upgrade on each of these versions but it doesn't work.

I'm obviously still missing something. Do I need to find out the Hardware Rev. of this unit?

drew7721, 2024-05-04:
I'm running in the same issue. Here are the STR: 
 - Latest original firmware 1.1.x (the one with the annoying persistent setup wizard and super complex password requirements) - **not on 1.2.x but the latest 1.1.x)
 - I tried the `initramfs-recovery-installer.itb` _v1.1.1) - did not work ❌ 
 - Then I tried the `initramfs-recovery-installer-signed.itb` _v1.1.1_ - it worked ✅ (since it seems latest original firmware on v1.1.x requires signed - linksys does not care about semantic versioning)
 - then I installed `openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb` from release _v1.1.1_
   - at this point Luci UI is gone but I have DHCP and SSH working
   - I SSH in and seem to have two UBI mounts
```
root@OpenWrt:/tmp# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 07f80000 00020000 "ubi"
```
uhttpd not running, no Luci
```
root@OpenWrt:/etc/sysctl.d# netstat -anpt
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 192.168.1.1:53          0.0.0.0:*               LISTEN      2285/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      2285/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1442/dropbear
tcp        0    196 192.168.1.1:22          192.168.1.101:50436     ESTABLISHED 2559/dropbear
tcp        0      0 fdf0:bd6b:a693::1:53    :::*                    LISTEN      2285/dnsmasq
tcp        0      0 ::1:53                  :::*                    LISTEN      2285/dnsmasq
tcp        0      0 fe80::daec:5eff:fe44:4e9c:53 :::*                    LISTEN      2285/dnsmasq
tcp        0      0 fe80::daec:5eff:fe44:4e9c:53 :::*                    LISTEN      2285/dnsmasq
tcp        0      0 :::22                   :::*                    LISTEN      1442/dropbear
root@OpenWrt:/etc/sysctl.d# ubus call system board
{
        "kernel": "6.1.79",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Linksys E8450 (UBI)",
        "board_name": "linksys,e8450-ubi",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r25361-a86ff06d2e",
                "target": "mediatek/mt7622",
                "description": "OpenWrt SNAPSHOT r25361-a86ff06d2e"
        }
}
```
I think this is where I messed up...
> I should have gotten this one: https://firmware-selector.openwrt.org/?version=23.05.3&target=mediatek%2Fmt7622&id=linksys_e8450-ubi and maybe use the "Attended system upgrade"??

Instead, I downloaded the latest version from here:
https://firmware-selector.openwrt.org/?version=23.05.3&target=mediatek%2Fmt7622&id=linksys_e8450
This is **not the UBI version** 🤦🏻‍♂️ 

I installed the non UBI version by using ssh (scp did not work) and `sysupgrade`.

Now I have Luci working but it's under `initramfs` mode..

I tried installing the correct UBI version through Luci UI but still stuck in `initramfs` mode. 

Now I get this with `cat /proc/mtd` and `ubus call system board`
```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "Preloader"
mtd1: 00040000 00020000 "ATF"
mtd2: 00080000 00020000 "u-boot"
mtd3: 00080000 00020000 "u-boot-env"
mtd4: 00100000 00020000 "factory"
mtd5: 00020000 00020000 "devinfo"
mtd6: 00020000 00020000 "senv"
mtd7: 00020000 00020000 "bootseq"
mtd8: 01e00000 00020000 "firmware1"
[... 155 lines cut ...]
<6>[  121.905562] device lan3 entered promiscuous mode
<6>[  121.916796] mt7530-mdio mdio-bus:00 lan4: configuring for phy/gmii link mode
<6>[  121.927061] br-lan: port 4(lan4) entered blocking state
<6>[  121.932324] br-lan: port 4(lan4) entered disabled state
<6>[  121.939326] device lan4 entered promiscuous mode
<6>[  121.946952] br-lan: port 4(lan4) entered blocking state
<6>[  121.952222] br-lan: port 4(lan4) entered forwarding state
<6>[  122.451497] br-lan: port 4(lan4) entered disabled state
<6>[  125.107018] mt7530-mdio mdio-bus:00 lan4: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[  125.115059] br-lan: port 4(lan4) entered blocking state
<6>[  125.120297] br-lan: port 4(lan4) entered forwarding state
<4>[  264.214150] mtdblock: MTD device 'mfg' is NAND, please consider using UBI block devices instead.
<6>[ 1746.476788] sysrq: Trigger a crash
<0>[ 1746.480222] Kernel panic - not syncing: sysrq triggered crash
<2>[ 1746.485968] SMP: stopping secondary CPUs
<0>[ 1746.489903] Kernel Offset: disabled
<0>[ 1746.493388] CPU features: 0x0,00001000,00000802
<0>[ 1746.497921] Memory Limit: none
```

I've tried deleting and rebooting and still nothing.

Is there anything that can be done to fix this? I've ran out of clues and knowledge on how to fix this. 

### 🆘 Help!!

mrkiko, 2024-05-04:
There is a kind of barrier between 32.x and snapshot.
In partcular, the installer has been modified in the aim of more reliability.
It's won0t help probably to repat some informations here - it makes more sense to look at the README.md file.

drew7721, 2024-05-09:
OK, I've managed to fix this. These are the steps:
 - ssh into router
 - run `echo c > /proc/sysrq-trigger`
 - Router reboots (you will no be in the UBI mode)
 - Upload firmware from 1.1.1 `openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb`
 - Router reboots (no-luci) 
 - From https://firmware-selector.openwrt.org/ download UBI version Snapshot but **use custom build** to get luci
 - Download sysupgrade file (name it `sysupgrade.itb` for short)
 - Use scp to upload file to router `ssh root@192.168.1.1 "cat > /tmp/sysupgrade.itb" < ./sysupgrade.itb`
 - ssh into router and run `sysupgrade -v /tmp/sysupgrade.itb`
 - Router restarts
 - Login to Luci and you're good to go!

closed 2025-07-04 by dangowrt