dangowrt/owrt-ubi-installer #180: Downgrade from v1.1.1 to  v1.0.3

PovilasID, 2024-03-27:
Hey,

I installed v1.1.1 and everything worked fine but I need Wireguard that was incompatible with kernel version.
Can I downgrade from v1.1.1 to v1.0.3 by just upload sysupgrade from luci or should I just wait for stable?

P.S.  I forced Wireguard install despite incompatibility so it now does not restart on `reboot` command which is admittedly inconvinient.

mrkiko, 2024-05-04:
Hi!
The installer is only a method to install OpenWrt, it's not meant as a version of OpenWrt you should use.
You're just better grab a snapshot from downloads.openwrt.org (be careful to select the ubi layout of course), and install all the packages you need.
This has the inconvenience that some packages might be installable for a limited amount of time. This can be fixed by installing them all ASAP or build your own snapshot and include the packages you need or keep them around archived somwehre.

PovilasID, 2024-05-18:
Update: So... I did not do what you told me and now I think it is bricked 🤣 
I was running your latest version for a few months minimal stability issues and this morning it looks like it died. No leds come no wifi nothing. I will do more diagnostics later but it looks like it is bricked now.
I was still ironing out the settings and just did not find the time to install latest snapshot.

dangowrt, 2025-07-04:
Most likely you will need to connect to the internal 3.3V-level serial port, and you may need to use `mtk_uartboot` to fix the issue. Re-open this issue should you still need help recovering the device.

PovilasID, 2025-07-04:
I was able to recover the device using USB to TTL process.