dangowrt/owrt-ubi-installer #186: I need help with some issues. My device light is off.

wangpengju00, 2024-06-12:
My device has switched to the UBI layout, but I installed a self-built system image before (the recovery on this page), and then I mistakenly updated the openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb (1.12) on this page under the sysupgrade page (only installed this), and now the device indicator light is not on, so I checked ttl output, it says "BL2: Failed to load image id 3" just like this: https://forum.openwrt.org/t/need-help-with-a-bricked-linksys-e8450/144836, I don't know what went wrong, and how I should solve the problem
[image]

wangpengju00, 2024-06-12:
[image]
[image]
I only have one such device that can connect to the e8450 and display this data, can I use it to solve my problem?

closed 2024-06-12 by wangpengju00

dangowrt, 2024-06-12:
Yes, this adapter should work. Connect only GND to GND, TX to RX and RX to TX.

wangpengju00, 2024-06-13:
OK, but after I use this adapter to connect to the motherboard, it only displays the following content, and there is no operable part. How can I use this adapter to solve my problem? Thank you.

F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0400 0041 [0000]
G0: 1190 0000
T0: 0000 02AF [000F]
Jump to BL

NOTICE:  BL2: v2.4(release):OpenWrt v2021-05-08-d2c75b21-3 (mt7622-snand-1ddr)
NOTICE:  BL2: Built : 02:55:34, Sep  3 2022
NOTICE:  SPI-NAND: FM35Q1GA (128MB)
ERROR:   BL2: Failed to load image id 3 (-2)

wangpengju00, 2024-06-13:
> Yes, this adapter should work. Connect only GND to GND, TX to RX and RX to TX.

OK, but after I use this adapter to connect to the motherboard, it only displays the following content, and there is no operable part. How can I use this adapter to solve my problem? Thank you.

F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0400 0041 [0000]
G0: 1190 0000
T0: 0000 02AF [000F]
Jump to BL

NOTICE: BL2: v2.4(release):OpenWrt v2021-05-08-d2c75b21-3 (mt7622-snand-1ddr)
NOTICE: BL2: Built : 02:55:34, Sep 3 2022
NOTICE: SPI-NAND: FM35Q1GA (128MB)
ERROR: BL2: Failed to load image id 3 (-2)

wangpengju00, 2024-06-13:
This is the latest content displayed on my computer
[image]

mrkiko, 2024-06-13:
> > Yes, this adapter should work. Connect only GND to GND, TX to RX and RX to TX.
> 
> OK, but after I use this adapter to connect to the motherboard, it only displays the following content, and there is no operable part. How can I use this adapter to solve my problem? Thank you.
> [image]

Hi!

you will probably need
mtk_uartboot
search for it around in the openwrt forum and/or maybe the issues (even closed) of this repo.

user8446, 2024-07-04:
Instructions to recover from the "OpenWrt Kiss of Death":

https://openwrt.org/toh/linksys/e8450#recovery_from_openwrt_kiss_of_death_okd