dangowrt/owrt-ubi-installer #187: Authentication loop

dalama32, 2024-06-19:
I have a brand new EA8450 AX3200 out of the box firmware 1.0.01
When I get to step 6 (authentication required) and enter the password "admin", it appears to accept the password then immediately asks for it again (endless loop)?

Any ideas?

randomouscrap98, 2024-10-03:
I was ale to get past this by using an old browser or alternative browser (in my case, I used [seamonkey](https://www.seamonkey-project.org/). See: https://www.reddit.com/r/Linksys/comments/1fi15p9/linksys_e8450_cant_access_web_interface_it/

danmarce, 2025-03-18:
Yeah, I had the same issue, first time and I have done a few of these routers. I had the same instinct. Seamonkey worked.