dangowrt/owrt-ubi-installer #188: Missing instructions on how to upgrade TF-A/U-Boot to newer versions.

hrw, 2024-06-26:
I installed OpenWRT on E8450 in 2021. Used installer 0.5.3 (looking at TF-A 2.4 + U-Boot 2021.04 versions).

Device refuses to upgrade from 23.05.2 -> 23.05.3 and I started to wonder do I need to update bootloaders.

But your docs say "vendor FW -> OpenWRT" and vice-versa only.

Please add instructions on how to update bootloaders.

justinkb, 2024-07-26:
I guess we just need to revert to original firmware and then install openwrt again

hrw, 2024-07-26:
There should be also a way to do it without going to vendor firmware.

justinkb, 2024-07-26:
Yes, I've just found it. You can find the details in the comments on the
commit from the dev branch that introduces the fip on ubi change for this
device

hrw, 2024-07-26:
You found it then please post a link to change.

Nit everyone has a time or skills to find such data. And there will be users needing it

justinkb, 2024-07-26:
https://github.com/openwrt/openwrt/commit/6aec3c7b5bf5e5a999a12121dfa71963afb6f003#commitcomment-139151946

https://github.com/openwrt/openwrt/commit/6aec3c7b5bf5e5a999a12121dfa71963afb6f003#commitcomment-139240065

hrw, 2024-12-03:
Followed instructions, OpenWRT 24.10-rc1 works on my device.

Thanks!

closed 2024-12-03 by hrw