dangowrt/owrt-ubi-installer #189: The official version v22.03.7 is not available under installer version 1.0.3

zow2023, 2024-08-07:
After flashing the firmware, go directly back to the installer interface

zow2023, 2024-09-13:
Oops#1 Part1
<6>[    0.105524] CPU: All CPU(s) started at EL2
<6>[    0.109637] alternatives: patching kernel code
<6>[    0.117635] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.127536] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
<6>[    0.134430] pinctrl core: initialized pinctrl subsystem
<6>[    0.140994] NET: Registered protocol family 16
<6>[    0.145875] DMA: preallocated 128 KiB GFP_KERNEL pool for atomic allocations
<6>[    0.153010] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations
<6>[    0.160798] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA32 pool for atomic allocations
<6>[    0.169110] thermal_sys: Registered thermal governor 'fair_share'
<6>[    0.169117] thermal_sys: Registered thermal governor 'bang_bang'
<6>[    0.175232] thermal_sys: Registered thermal governor 'step_wise'
<6>[    0.181298] thermal_sys: Registered thermal governor 'user_space'
<6>[    0.187610] ASID allocator initialised with 65536 entries
<6>[    0.199682] pstore: Registered ramoops as persistent store backend
<6>[    0.205919] ramoops: using 0x10000@0x42ff0000, ecc: 0
<3>[    0.221011] Cannot find regmap for /clock-controller@10209000: -22
<1>[    0.227278] Unable to handle kernel paging request at virtual address fffffffffffffffe
<1>[    0.235224] Mem abort info:
<1>[    0.238027]   ESR = 0x96000005
<1>[    0.241086]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[    0.246411]   SET = 0, FnV = 0
<1>[    0.249471]   EA = 0, S1PTW = 0
<1>[    0.252615] Data abort info:
<1>[    0.255498]   ISV = 0, ISS = 0x00000005
<1>[    0.259340]   CM = 0, WnR = 0
<1>[    0.262315] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000044990000
<1>[    0.269036] [fffffffffffffffe] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[    0.277777] Internal error: Oops: 0000000096000005 [#1] SMP
<7>[    0.283365] Modules linked in:
<7>[    0.286427] CPU: 0 PID: 1 Comm: swapper/0 Tainted: G S                5.10.221 #0
<7>[    0.293934] Hardware name: Linksys E8450 (UBI) (DT)
<7>[    0.298826] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
<7>[    0.304851] pc : 0xffffffc010455e44
<7>[    0.308345] lr : 0xffffffc01045eb20
<7>[    0.311838] sp : ffffffc010b5bb20
<7>[    0.315157] x29: ffffffc010b5bb20 x28: 0000000000000000 
<7>[    0.320485] x27: ffffffc0109a03c8 x26: ffffffc010abd378 
<7>[    0.325814] x25: ffffffc0109885e8 x24: 0000000000000000 
<7>[    0.331142] x23: ffffffc010ab7f08 x22: 0000000000000000 
<7>[    0.336471] x21: ffffff801fee5fa8 x20: fffffffffffffffe 
<7>[    0.341799] x19: 0000000000000000 x18: 000000000000004e 
<7>[    0.347127] x17: 000000000000001c x16: 0000000049e881dd 
<7>[    0.352456] x15: ffffffc010a29b60 x14: 00000000000000ea 
<7>[    0.357784] x13: 000000000000004e x12: ffffffc010b5b7f8 
<7>[    0.363112] x11: ffffffc010a81b60 x10: 00000000fffff000 
<7>[    0.368440] x9 : ffffffc010a81b60 x8 : 0000000000000000 
<7>[    0.373769] x7 : ffffffc010a29b60 x6 : 00000000ffffefff 
<7>[    0.379097] x5 : 0000000000000000 x4 : 0000000000000000 
<7>[    0.384425] x3 : ffffff8000050000 x2 : 0000000000000000 
<7>[    0.389753] x1 : ffffffc010b2c418 x0 : fffffffffffffffe 
<7>[    0.395082] Call trace:
<7>[    0.397530]  0xffffffc010455e44
<7>[    0.400674]  0xffffffc01045eb20
<7>[    0.403819]  0xffffffc01045e970
<7>[    0.406963]  0xffffffc0104a48b0
<7>[    0.410108]  0xffffffc0104a246c
<7>[    0.413252]  0xffffffc0104a2ba4
<7>[    0.416396]  0xffffffc0104a2e10
<7>[    0.419541]  0xffffffc0104a01f8
<7>[    0.422685]  0xffffffc0104a28c4
<7>[    0.425829]  0xffffffc0104a2e80
<7>[    0.428974]  0xffffffc0104a14d8
<7>[    0.432118]  0xffffffc01049ec38
<7>[    0.435262]  0xffffffc0109c1694
<7>[    0.438407]  0xffffffc010004b70
<7>[    0.441551]  0xffffffc0109a1024
<7>[    0.444696]  0xffffffc010810394
<7>[    0.447840]  0xffffffc010005f78
[... 50 lines cut ...]
<7>[    0.379097] x5 : 0000000000000000 x4 : 0000000000000000 
<7>[    0.384425] x3 : ffffff8000050000 x2 : 0000000000000000 
<7>[    0.389753] x1 : ffffffc010b2c418 x0 : fffffffffffffffe 
<7>[    0.395082] Call trace:
<7>[    0.397530]  0xffffffc010455e44
<7>[    0.400674]  0xffffffc01045eb20
<7>[    0.403819]  0xffffffc01045e970
<7>[    0.406963]  0xffffffc0104a48b0
<7>[    0.410108]  0xffffffc0104a246c
<7>[    0.413252]  0xffffffc0104a2ba4
<7>[    0.416396]  0xffffffc0104a2e10
<7>[    0.419541]  0xffffffc0104a01f8
<7>[    0.422685]  0xffffffc0104a28c4
<7>[    0.425829]  0xffffffc0104a2e80
<7>[    0.428974]  0xffffffc0104a14d8
<7>[    0.432118]  0xffffffc01049ec38
<7>[    0.435262]  0xffffffc0109c1694
<7>[    0.438407]  0xffffffc010004b70
<7>[    0.441551]  0xffffffc0109a1024
<7>[    0.444696]  0xffffffc010810394
<7>[    0.447840]  0xffffffc010005f78
<0>[    0.450990] Code: 910003fd f9000bf3 52800013 b40000e0 (f9400013) 
<4>[    0.457115] ---[ end trace dafcd3ca2cdcc63b ]---
<0>[    0.462346] Kernel panic - not syncing: Oops: Fatal exception
<2>[    0.468111] SMP: stopping secondary CPUs

zow2023, 2024-09-13:
Following the previous prompts, deleting the above two files in the corresponding locations does not solve the problem

tabcat, 2024-09-26:
I have the same logs after trying to upgrade to v22.03.7

dangowrt, 2024-09-26:
Please report this as a bug to github.com/openwrt/openwrt as it is not related to the installer itself, but rather seems to be a kernel issues of OpenWrt 22.03.7.

tabcat, 2024-09-26:
https://github.com/openwrt/openwrt/issues/16515

closed 2024-09-27 by zow2023