dangowrt/owrt-ubi-installer #190: Deletion of bl2-for-mtk_uartboot.bin

petrij98, 2024-08-16:
I can no longer recover my bricked Belkin RT3200 because all instances of this essential file for mtk_uartboot have been removed. Using this file is critical in the official recovery guide on OpenWRT's website.

dangowrt, 2024-08-16:
Images for `mtk_uartboot` are now built for all newer MediaTek platforms in the regular OpenWrt build system. I've opened a PR https://github.com/openwrt/openwrt/pull/16178 which suggests to do the same also for MT7622.

jaws35smith, 2024-08-17:
@dangowrt so does that mean the recovery procedures are now incorrect? @petrij98 I would recommend using wayback machine to grab the .bin in the mean time until the recovery procedures are updated.

dangowrt, 2024-08-17:
You can grab the file here: https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/mt7622-ram-1ddr-bl2.bin
Instructions will need to be updated.

closed 2024-08-26 by dangowrt