dangowrt/owrt-ubi-installer #191: Why was release tag 1.0.3 removed from releases section

Eden33, 2024-08-26:
Hi, 

2 months ago I downloaded and installed that firmware on my router:
https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v1.0.3

I'm sure that that time it was not marked as "Pre-Release" and also quite sure it was highlighted that it is the latest stable version (otherwise I would not have flashed it).

Checking now the release section (https://github.com/dangowrt/owrt-ubi-installer/releases) this release is not present anymore.

May I ask you why? ... so I can make a decision. Thanks

Eden33, 2024-08-26:
OK, in case I am not mistaken 1.0.3 up to 1.1.2 occasionally subject to face  https://openwrt.org/toh/linksys/e8450#brief_summary_of_okd. However, is there an additional reason why 1.0.3 was removed from releases page whereas 1.1.1 and 1.1.2 remains to be listed?

@dangowrt 
Any recommendations how to proceed wit a device (Belkin RT3200) flashed with 1.0.3?

Thank you for your reply and corrections (in case I summed something up incorrect).

Eden33, 2024-08-26:
@dangowrt 
To be more precise about my upgrade path till date because I think it may matters in case you find time commenting on it:

**Initially from Belkin RT3200 stock firmware I flashed (this was in 2023):**

openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
openwrt-22.03.3-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

After that my device was on OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598

**2 months ago I flashed:**

openwrt-23.05.0-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
(it was present that time under the following that link: https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v1.0.3)

Since that time my device is on 
OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-23.236.53405-fc638c8

dangowrt, 2024-08-26:
> May I ask you why? ... so I can make a decision. Thanks

Yes, they were removed because of OKD, users in the fourm [suggested](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/5780?u=daniel) removing affected releases.

> However, is there an additional reason why 1.0.3 was removed from releases page whereas 1.1.1 and 1.1.2 remains to be listed?

Later I figured that I could just remove the binaries and leave the release notes (though there isn't much worth in having a source download, because the source download for older releases won't even work to generate working binaries by now, because it depends on content on downloads.openwrt.org which has changed in the meantime).

From your description I believe that your device could currently be affected by OKD. See [this post](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/5717?u=daniel) to find out for sure and what to do.

Eden33, 2024-08-26:
Thank you for your time and the helpful reference provided. Lucky me, the BL2 is exactly on the 2.4 version outlined in the reference.

closed 2025-07-04 by dangowrt