dangowrt/owrt-ubi-installer #193: Installed via release v1.0.2, is it safe to update OpenWrt to latest stable using auc/luci-app-attendedsysupgrade?

suarezjulian, 2024-09-12:
Hi, 

I stayed away from the removed installer releases, and installed OpenWrt using release v1.0.2, but seeing the issues and the notes on the newer releases I'm unsure if it is safe to install a newer version of OpenWrt. 

I guess I'm not sure if the installer releases on this repo are tied to OpenWrt releases or not.

I apologize if this is obvious or clear for other people but I couldn't find the answer searching on this repo.

Thanks

dangowrt, 2024-09-22:
You may safely upgrade to latest 23.05.x release or even 23.05-SNAPSHOT. For upcoming 24.10 release or SNAPSHOT builds you will need to run installer v1.1.3 or later.

suarezjulian, 2024-09-22:
Thanks @dangowrt

closed 2024-09-22 by suarezjulian