dangowrt/owrt-ubi-installer #197: Help! factory partition issue

widewing, 2024-10-05:
Hi, yesterday I tried to use the 1.1.3 installer to update the UBI layout, but experienced multiple issues, cannot enable 5ghz, the MAC address is random, etc. So I decided to rollback the older version, by using 1.0.2 recovery image to write the backup mtd files.

I initially thought it rolled back successfully, however I noticed the Ethernet MAC is still randomly generated. Wifi MAC is correct though. `fw_printenv` shows the correct factory mac. So I dig into the issue and it looks like it cannot read the MAC from factory partition.

Then I found the backup file for the factory (mtd2) is 128KB while the factory partition in the system is 1MB! Then I checked the Wiki, 1MB should be the correct size. All other backup partitions are correct size (bl2: 512KB, fip: 1280KB, ubi: 128000KB).

It seems only affects the MAC address for now. I can workaround it by configuring the MAC manually, but I'm wondering 1. what could be the possible reason? 2. is it possible to be fixed? 3. are there any other potential problems?

dangowrt, 2024-10-05:
It's kinda impossible to know what happened, and especially why the 1.1.3 installer didn't work for you, without any further context.

It is important to note that the 1.1.x installer splits the flash into only two parts, `bl2` and `ubi`, and hence also the backup files created by the 1.1.x installer cover all relevant parts only within those two partitions (`mtd0` and `mtd1`). The fact that you state you got 4 files means you are dealing with backup files which have been created by the old 1.0.x installer, right?

Please share the backup files, so I can take a look and understand what happened. Maybe the device got a bad block right in the middle of the `factory` partition, so calibration data is in place, but read may abort and hence the part of the MAC addresses could be missing (there is a quite large unused gap in between calibration data and MAC addresses, a bad block there wouldn't affect normal operation, but it may confuse the installer and cause problems similar to what you are describing).

widewing, 2024-10-05:
Thanks for replying! Yes I made backup with old installer (I guess should from 0.x version). I understand the MAC address is after the 128KB (I checked the DTS file, should be 0x7fff4 which is ~514KB). 

Actually I backuped twice: 3 years ago after I flashed openwrt, and yesterday before upgrading new installer. They both have same mtd2 (factory) size of 128KB and same content. Not sure how to upload but I can share the screenshot
[image]
and
[image]

Unfortunately I didn't make backup after the new installer (the two partitions scheme)

Also, after flaching 1.0.2 recovery and restoring the backup files, it can successfully generate the full 1MB backup file for factory partition now.
[image]
Does it mean it's not because of bad blocks in the middle of factory partition? Maybe some issue with earlier version of the installer?

dangowrt, 2025-07-04:
Please upload the full backup files somewhere and share them with me *privately* (as they may contain sensitive data such as WiFi credentials). We can then figure out what happened to the `factory` partition and I can help you to fully recover the device.

closed 2025-07-04 by widewing