dangowrt/owrt-ubi-installer #201: Public key changes?

RSpliet, 2024-11-09:
I recently tried to re-generate some images (for full discloser: using an older version of the script), and ran into the issue of the script bailing early with
`gpg: Can't check signature: No public key`

Looking at the signatures, it looks like the sha256 key files as downloaded today no longer contain the deprecated public key with signature 0xCD84BCED626471F1 . I thus tracked the issue down th the following line:

https://github.com/dangowrt/owrt-ubi-installer/blob/e0f62de5357158e444c9b24451ae604ee27fa8b2/build_installer.sh#L8C1-L8C33

Replacing the key with 0x1D53D1877742E911 (Universal GPG signing key from Nitrokey 3A key) resolves the issue.

dangowrt, 2024-11-09:
Yes, the build key has changed. Can you open a PR with that change please?

RSpliet, 2024-12-01:
Ping! Opened a merge request a few weeks ago, just wanted to draw your attention to it in case it fell off the radar.

closed 2024-12-03 by RSpliet