dangowrt/owrt-ubi-installer #204: opkg: No such file or directory

r-gudmk, 2024-11-16:
After fixing the public key issue the script fails with:

./build_installer.sh: line 198: /tmp/owrt-ubi-installer/openwrt-ib/staging_dir/host/bin/opkg: No such file or directory

knazarov, 2025-01-08:
It seems like this is because OpenWrt is moving away from opkg (see [this topic for some details](https://forum.openwrt.org/t/opkg-not-found-after-upgrading-to-r27998-591272d197/214723/9)).

I've succeeded with adding opkg by hacking its installation back to owrt-ubi-installer's Containerfile. But it ended up breaking down the line because the snapshot release of OpenWrt which this ubi installer requires, no longer ships opkg.conf.

I guess, the right course of action will be to migrate to APK.

dangowrt, 2025-01-09:
Yes, for 25.xx release we will have to migrate to APK, the 24.10.0 release is still based on OPKG and I will release the next version of the installer based on 24.10.0 once it gets released.

knazarov, 2025-01-09:
@dangowrt so, if we pin the version of openwrt to 24.10.0 instead of "snapshot" then the issue will be fixed for now?

dangowrt, 2025-01-09:
> so, if we pin the version of openwrt to 24.10.0 instead of "snapshot" then the issue will be fixed for now?

Yes, we can use 24.10.0-rc5 for now, and then 24.10.0 once it gets released.

knazarov, 2025-01-09:
I've contributed a pull request for anyone interested in fixing this problem: https://github.com/dangowrt/owrt-ubi-installer/pull/211