dangowrt/owrt-ubi-installer #207: Question about revert back to vendor's firmware

llkpersonal, 2024-11-26:
Greetings!

I recently using v1.1.3 installer to flash my E8450, therefore it has only two files in the `boot_backup` folder. I am wondering in this case how can I go back to the vendor's firmware as there are four files listed in the readme manual.

I assume it should be:
```
ubidetach -d 0
insmod mtd-rw i_want_a_brick=1
mtd write /tmp/mtd0 /dev/mtd0
mtd write /tmp/mtd1 /dev/mtd1
```

followed by
```
mtd -p 0x200000 write /tmp/FW_E8450_1.0.01.101415_prod.img /dev/mtd3
```

Is it correct?

john-clark, 2024-11-27:
where do you get /tmp/mtd0 /tmp/mtd1

dalutou, 2025-02-13:
```
mtd -p 0x200000 write /tmp/FW_E8450_1.0.01.101415_prod.img /dev/mtd3
```
It should not work, because there is no `mtd3` after  v1.1.3 installer.

I think we should write to mtd1, but I don't know the offset value.

```
mtd -p ##OFFSET_VALUE## write /tmp/FW_E8450_1.0.01.101415_prod.img /dev/mtd1
```

Maybe [Restoring the vendor/official firmware](https://github.com/dangowrt/owrt-ubi-installer?tab=readme-ov-file#restoring-the-vendorofficial-firmware) in `README.md` should be updated for the mtd backups by v1.1.x installer.

@dangowrt

dalutou, 2025-02-14:
```
mtd -p 0x200000 write /tmp/FW_E8450_1.0.01.101415_prod.img /dev/mtd3
```
It should not work, because there is no `mtd3` since v1.1.x installer.

I think we should do it this way (I haven't tested the following operations)

```
ubidetach -d 0
insmod mtd-rw i_want_a_brick=1
mtd write /tmp/mtd0 /dev/mtd0
mtd write /tmp/mtd1 /dev/mtd1
reboot
```

Then you get the vendor u-boot. Connect to the internal serial port, flash the vendor firmware image through the u-boot menu.

Reboot the device and you have restore the vendor firmware.

llkpersonal, 2025-02-14:
I have successfully restore my vendor's firmware, by the operations I talked about. Thank you all!

closed 2025-02-14 by llkpersonal