dangowrt/owrt-ubi-installer #213: Failed to update with new installer

smileys29, 2025-02-08:
I tried to update one of my RT3200's and ran into a snag.  Here's my fw_printenv:

`boot_default=if env exists flag_recover ; then else run bootcmd ; fi ; run boot_recovery ; setenv replacevol 1 ; run boot_tftp_forever
boot_first=if button reset ; then led $bootled_rec on ; run boot_tftp_recovery ; setenv flag_recover 1 ; run boot_default ; fi ; bootmenu
boot_production=led $bootled_pwr on ; run ubi_read_production && bootm $loadaddr#$bootconf ; led $bootled_pwr off
boot_recovery=led $bootled_rec on ; run ubi_read_recovery ; bootm $loadaddr#$bootconf ; ubi remove recovery ; led $bootled_rec off
boot_serial_write_bl2=loadx $loadaddr 115200 && run snand_write_bl2
boot_serial_write_fip=loadx $loadaddr 115200 && run ubi_write_fip
boot_tftp=tftpboot $loadaddr $bootfile && bootm $loadaddr#$bootconf
boot_tftp_forever=led $bootled_rec on ; while true ; do run boot_tftp_recovery ; sleep 1 ; done
boot_tftp_production=tftpboot $loadaddr $bootfile_upg && iminfo $loadaddr && ubi part ubi && run ubi_write_production ubi_prepare_rootfs ; if env exists noboot ; then else bootm $loadaddr#$bootconf ; fi
boot_tftp_recovery=tftpboot $loadaddr $bootfile && iminfo $loadaddr && ubi part ubi && run ubi_write_recovery ; if env exists noboot ; then else bootm $loadaddr#$bootconf ; fi
boot_tftp_write_bl2=tftpboot $loadaddr $bootfile_bl2 && run snand_write_bl2
boot_tftp_write_fip=tftpboot $loadaddr $bootfile_fip && run ubi_write_fip
boot_ubi=ubi part ubi && run boot_production ; run boot_recovery
bootcmd=if pstore check ; then run boot_recovery ; else run boot_ubi ; fi
bootconf=config-1
bootdelay=3
bootfile=openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb
bootfile_bl2=openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin
bootfile_fip=openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip
bootfile_upg=openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
bootled_pwr=power:blue
bootled_rec=inet:orange on
bootmenu_0=Run default boot command.=run boot_default
bootmenu_1=Boot system via TFTP.=run boot_tftp ; run bootmenu_confirm_return
bootmenu_2=Boot production system from flash.=run boot_production ; run bootmenu_confirm_return
bootmenu_3=Boot recovery system from flash.=run boot_recovery ; run bootmenu_confirm_return
bootmenu_4=Load production system via TFTP then write to flash.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_production ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
bootmenu_5=Load recovery system via TFTP then write to flash.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_recovery ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
bootmenu_6=Load BL31+U-Boot FIP via TFTP then write to flash.=run boot_tftp_write_fip ; run bootmenu_confirm_return
bootmenu_7=Load BL2 preloader via TFTP then write to flash.=run boot_tftp_write_bl2 ; run bootmenu_confirm_return
bootmenu_8=Reboot.=reset
bootmenu_9=Reset all settings to factory defaults.=run reset_factory ; reset
bootmenu_confirm_return=askenv - Press ENTER to return to menu ; bootmenu 60
bootmenu_default=0
bootmenu_delay=3
bootmenu_title=      ( ( ( OpenWrt ) ) )       U-Boot 2024.01-OpenWrt-r26536-c67d5189a4 (Jun 06 2024 - 09:47:59 +0000)
ethaddr=xx:xx:xx:xx:xx:xx
filesize=80000
ipaddr=192.168.1.1
loadaddr=0x48000000
reset_factory=mw $loadaddr 0xff 0x1f000 ; ubi part ubi ; ubi write $loadaddr ubootenv 0x1f000 ; ubi write $loadaddr ubootenv2 0x1f000 ; ubi remove rootfs_data
serverip=192.168.1.254
snand_write_bl2=mtd erase bl2 && mtd write spi-nand0 $loadaddr 0x0 0x20000 && mtd write spi-nand0 $loadaddr 0x20000 0x20000 && mtd write spi-nand0 $loadaddr 0x40000 0x20000 && mtd write spi-nand0 $loadaddr 0x60000 0x20000
ubi_prepare_rootfs=if ubi check rootfs_data ; then else if env exists rootfs_data_max ; then ubi create rootfs_data $rootfs_data_max dynamic || ubi create rootfs_data - dynamic ; else ubi create rootfs_data - dynamic ; fi ; fi
ubi_read_production=ubi read $loadaddr fit && iminfo $loadaddr && run ubi_prepare_rootfs
ubi_read_recovery=ubi check recovery && ubi read $loadaddr recovery
ubi_remove_rootfs=ubi check rootfs_data && ubi remove rootfs_data
ubi_write_fip=run ubi_remove_rootfs ; ubi check fip && ubi remove fip ; ubi create fip 0x200000 static ; ubi write $loadaddr fip 0x200000
ubi_write_production=ubi check fit && env exists replacevol && ubi remove fit ; if ubi check fit ; then else run ubi_remove_rootfs ; ubi create fit $filesize dynamic && ubi write $loadaddr fit $filesize ; fi
ubi_write_recovery=ubi check recovery && env exists replacevol && ubi remove recovery ; if ubi check recovery ; then else run ubi_remove_rootfs ; ubi create recovery $filesize dynamic && ubi write $loadaddr recovery $filesize ; fi
ver=U-Boot 2024.01-OpenWrt-r26536-c67d5189a4 (Jun 06 2024 - 09:47:59 +0000)
`

You'll see that the U-Boot is still showing a SNAPSHOT build.  When the OKD fix came out, i was having upgrade issues then, so i manually updated the preloader to get the OKD fix and moved on.  I was hoping i could get the U-Boot updated with this iteration of the installer.

After i tried the installer firmware for the 24.10.0 release you just made, it rebooted OK and went into recovery.  i updated the firmware to a new firmware i grabbed from the Attended Sysupgrade tool and rebooted.  the RT3200 rebooted and went into recovery again.  I found there was a pstore file:

`Panic#1 Part1
<7>[    2.113878] pci_bus 0000:00: scanning bus
<6>[    2.118072] pci 0000:00:00.0: [14c3:3258] type 01 class 0x060400
<6>[    2.124260] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x1ffffffff 64bit pref]
<7>[    2.134663] pci_bus 0000:00: fixups for bus
<7>[    2.138896] pci 0000:00:00.0: scanning [bus 00-00] behind bridge, pass 0
<6>[    2.145595] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
<7>[    2.153673] pci 0000:00:00.0: scanning [bus 00-00] behind bridge, pass 1
<7>[    2.160807] pci_bus 0000:01: scanning bus
<6>[    2.164974] pci 0000:01:00.0: [14c3:7915] type 00 class 0x000280
<6>[    2.171168] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[... 69 lines cut ...]
<6>[    3.422632] xhci-mtk 1a0c0000.usb: xHCI Host Controller
<6>[    3.427891] xhci-mtk 1a0c0000.usb: new USB bus registered, assigned bus number 1
<6>[    3.436965] xhci-mtk 1a0c0000.usb: hcc params 0x01403198 hci version 0x96 quirks 0x0000000000200010
<6>[    3.446076] xhci-mtk 1a0c0000.usb: irq 133, io mem 0x1a0c0000
<6>[    3.451931] xhci-mtk 1a0c0000.usb: xHCI Host Controller
<6>[    3.457159] xhci-mtk 1a0c0000.usb: new USB bus registered, assigned bus number 2
<6>[    3.464566] xhci-mtk 1a0c0000.usb: Host supports USB 3.0 SuperSpeed
<6>[    3.471330] hub 1-0:1.0: USB hub found
<6>[    3.475146] hub 1-0:1.0: 2 ports detected
<6>[    3.479690] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
<6>[    3.488314] hub 2-0:1.0: USB hub found
<6>[    3.492161] hub 2-0:1.0: 1 port detected
<14>[    3.501470] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
<14>[    3.519212] init: - preinit -
<15>[    4.972028] INSTALLER: cannot find Wi-Fi EEPROM data
<6>[    4.978316] sysrq: Trigger a crash
<0>[    4.983085] Kernel panic - not syncing: sysrq triggered crash
<2>[    4.990139] SMP: stopping secondary CPUs
<0>[    4.994140] Kernel Offset: disabled
<0>[    4.997620] CPU features: 0x0,00000004,00000000,0000400b
<0>[    5.002926] Memory Limit: none
`
I tied this a couple times with the same result.  If needed i can open the device to onto the serial connection, if there's diagnostic data i can grab from there.

Appreciate the help in advance

smileys29, 2025-02-08:
[pstore.txt](https://github.com/user-attachments/files/18716318/pstore.txt)
[fw_printenv.txt](https://github.com/user-attachments/files/18716319/fw_printenv.txt)

dangowrt, 2025-02-08:
You cant update U-Boot builds which are everything-in-UBI by rerunning the installer, it will fail because it expects factory data in MTD partition rather than UBI volume. If you like you can update U-Boot using the serial console, but there is no need to do so.

dalutou, 2025-02-10:
My E8450 is running on OpenWrt v2024.10.0 stable(by owrt-1.1.3-installer)。 

I manually updated bl2(preloader) and uboot.fip .

```shell
opkg install kmod-mtd-rw 
insmod mtd-rw.ko i_want_a_brick=1
mtd -p 0x0 write /tmp/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-preloader.bin /dev/mtd0
mtd -p 0x20000 write /tmp/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-preloader.bin /dev/mtd0
mtd -p 0x40000 write /tmp/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-preloader.bin /dev/mtd0
mtd -p 0x60000 write /tmp/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-preloader.bin /dev/mtd0

ubiupdatevol /dev/ubi0_0 /tmp/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip
```
But `fw_printenv` show `ver=U-Boot 2024.01`.

After truncate two ubootenv volumes, fw-printenv show the right version `ver=U-Boot 2024.10`.
```shell
ubiupdatevol /dev/ubi0_2 -t
ubiupdatevol /dev/ubi0_3 -t
```

unmesh59, 2025-02-13:
I too tried updating OpenWrt from 23.05 to 2024.10.0 via v1.1.3-installer using the instructions for retaining config by modifying the backup file. Unfortunately, the OpenWRT sysupgrade file appears to install but the box keeps rebooting into firmware install mode.

How do I debug this?

dangowrt, 2025-02-13:
> How do I debug this?

First step: check (and share, if any) the content of `/sys/fs/pstore/*`.

unmesh59, 2025-02-13:
@dangowrt 

In preparation for sharing the information you requested, I happened to completely power cycle the box and it booted into OpenWRT!

dangowrt, 2025-02-13:
pstore being empty and the device booting into production firmware is the result of removing power. It would have been interesting to know what was there, but now it's gone...

closed 2025-02-13 by dangowrt