dangowrt/owrt-ubi-installer #215: New Linksys E7350 Won't Accept UBI Firmware

andspenrob, 2025-02-13:
Current firmware version: 1.1.00.032 (unsigned) installed from [Linksys Support](https://support.linksys.com/kb/article/3776-en/).

When attempting to upgrade firmware to unsigned recovery-installer.itb I get "The image file is incorrect."

[image]

I noticed that the official Linksys Support firmware now has the extension .bin. Have tried renaming recovery-installer.itb to recovery-installer.bin. Router will then accept and attempt the upgrade but upgrade fails and falls back.

Have tried downgrading, have tried signed recovery installer - in all cases, need to rename to *.bin to avoid "The image file is incorrect." No installers succeed.

Have not updated to "signed" official firmware, since that is a one-way street to my understanding. Any ideas?

dangowrt, 2025-02-13:
I'm not sure this version of the stock firmware accepts unsigned upgrade. Did you try with the signed installer file?
Also there are problems with some browsers...

andspenrob, 2025-02-13:
Have tried with Firefox and Chrome... The upgrade from 1.0 to 1.1 unsigned stock firmware worked fine.

I tried both of these with an error on firmware upgrade:

- openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
- openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb 

To get the form on the upgrade page to accept them, though, I have to change the filename extension to .bin... That seems new compared to other prior attempts... They still fail out, though, so I'm wondering if there is some other new check in place on the stock firmware?

dangowrt, 2025-02-13:
I noticed that the firmware doesn't accept the file if the total length of the *local* path is too long. Ie. try to move the file to the root directory of eg. a pendrive and rename it to a short name, that should help in that case.

andspenrob, 2025-02-13:
I appreciate the suggestion! I did try this, but it doesn't alter the behavior. Whether putting the renamed recovery-installer.itb files in root (C:\) or some other similarly small path, I still get the original "The image file is incorrect." error. That "C:\fakepath..." you see in the screenshot is the same whether the path I submit is short or long.

dangowrt, 2025-02-13:
If you also tried different browsers, the best I can imagine you can do is to upgrade to stock firmware 1.2 (signed), simply because I only test 1.0 and 1.2 stock firmware and 1.1 may have strange behavior I'm not aware of.

andspenrob, 2025-02-13:
I can try moving to stock signed firmware... Your versions are a bit confusing - is that from the Belkin version? These are the latest versions I see on the Linksys site for the E7350:

[image]

andspenrob, 2025-02-13:
Also: is there fundamentally any difference between a .img or .bin or .itb file? Is it just a renamed extension with the same format?

dangowrt, 2025-02-13:
Sorry, I only now see the obvious. You got E7350. That's not the same hardware and cannot run the installer at all. Supported here is only E8450 and RT3200.

andspenrob, 2025-02-13:
🤦‍♂️ They look exactly the same! (I previously had a RT3200 and was just retracing my steps) Thank you for catching my oversight! I really appreciate the support!

closed 2025-02-13 by andspenrob