dangowrt/owrt-ubi-installer #217: Can't upgrade Linksys E8450 (UBI)

fiskvik, 2025-02-18:
`Image check failed:

Tue Feb 18 10:53:07 CET 2025 upgrade: The device is supported, but this image is incompatible for sysupgrade based on the image version (1.0->2.0). Tue Feb 18 10:53:07 CET 2025 upgrade: SPI-NAND flash layout changes require bootloader update. Please run the UBI installer version 1.1.0+ (unsigned) first. Image check failed.`

The device is already UBI. What am I being asked to do? I can't go back to scratch, the configuration of this router has taken years to get just right. Would appreciate any help.

mrkiko, 2025-02-18:
Hello!!

First of all, I would recommend a backupo of your configuration, just in case...

May I ask you the OpenWrt version you are running and the output of

cat /proc/mtd

?

To what version are you trying to upgrade?

I am under the impression you are on the "old" UBI layout, and in that case you will need to run the unsigned installer again to switch to the new all-UBI layout.

fiskvik, 2025-02-18:
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 00140000 00020000 "fip"
mtd2: 00100000 00020000 "factory"
mtd3: 07d00000 00020000 "ubi"

Trying to upgrade from 23.05.5 to 24.10. How do I "run the unsigned installer again"? On the 'Backup / Flash Firmware' page on the router? or? I can't go back to scratch, I need to keep the package selection / configuration files on the router, I can't spend weeks recreating the config. That's why I tried to use Attended Sysupgrade.

Thank you for your help! :)

CrazyLemon, 2025-02-28:
There was a comment in release notes about this issue. More on that here https://openwrt.org/toh/linksys/e8450#upgrading_an_ubi_installation_to_new_releases_after_2024-02_includes_all_snapshots_2410-snapshots_24100-rcx_releases_and_all_releases_in_the_foreseable_future

bananos, 2025-04-19:
I've followed the [instruction on upgrading installer to 1.1.4](https://openwrt.org/toh/linksys/e8450#upgrading_an_ubi_installation_to_new_releases_after_2024-02_includes_all_snapshots_2410-snapshots_24100-rcx_releases_and_all_releases_in_the_foreseable_future) but it does not seem to help at all.   In the end I am stuck in initramfs mode.

My current config:

```
root@OpenWrt:/tmp# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

```
root@OpenWrt:/tmp# grep "(release)" /dev/mtd0ro | tail -1
v2.4(release):OpenWrt v2021-05-08-d2c75b21-3 (mt7622-snand-1ddr)
```

```
root@OpenWrt:/tmp# fw_printenv ver
ver=U-Boot 2023.07.02-OpenWrt-r24012-d8dd03c46f (Jul 15 2024 - 22:14:18 +0000)
```

STR:

Step 1
```bash
wget -O installer.itb "https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.1.4/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb"
sysupgrade -F installer.itb
reboot
```
router goes into initramfs mode

Step 2
```bash
wget -O sysupgrade.itb "https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.1.4/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb"
sysupgrade -v sysupgrade.itb 
Sat Apr 19 19:41:28 UTC 2025 upgrade: The device is supported, but this image is incompatible for sysupgrade based on the image version (1.0->2.0).
Sat Apr 19 19:41:28 UTC 2025 upgrade: SPI-NAND flash layout changes require bootloader update. Please run the UBI installer version 1.1.0+ (unsigned) first.
Image check failed.
```

In order to revert everything back and boot normally, here is what helps:
```
rm /sys/fs/pstore/*
reboot
```

tagesuhu, 2025-07-11:
I'm not sure if the issue I'm seeing is the same one or if I should open a different one, but I'm trying to upgrade from version 24.10.1 to a newer build without success and I'm seeing the same message on the console. Both the previous image and the current one are selfbuilt, but none have local patches and 'target/linux/mediatek/image/mt6722.mk' shows 'DEVICE_COMPAT_VERSION :=2.0' under Device/linksys_e8450-ubi, as expected.

IIRC, I've previously used the installer version 1.1.3 to upgrade from 23.05 to 24.10.

<pre>Fri Jul 11 19:49:57 -03 2025 upgrade: The device is supported, but this image is incompatible for sysupgrade based on the image version (1.0->2.0).
Fri Jul 11 19:49:57 -03 2025 upgrade: SPI-NAND flash layout changes require bootloader update. Please run the UBI installer version 1.1.0+ (unsigned) first.
Image check failed.</pre>

/etc/openwrt_release

<pre>DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.1'
DISTRIB_REVISION='r28597-0425664679'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 24.10.1 r28597-0425664679'
DISTRIB_TAINTS='no-all busybox'</pre>

/proc/mtd

<pre>dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 07f80000 00020000 "ubi"</pre>

/dev/mtd0ro

<pre>v2.10.0 (release):OpenWrt v2024.01.17~bacca82a-3 (mt7622-snand-ubi-1ddr)</pre>

<pre>$ fw_printenv ver
ver=U-Boot 2024.10-OpenWrt-r28427-6df0e3d02a (Feb 03 2025 - 23:09:37 +0000)</pre>

Nothing on /sys/fs/pstore

I did not run the installer yet again, as the device wiki warns against it... should I? If the only drawback is losing the previous working image, I can back it up.

Thanks in advance!

tagesuhu, 2025-07-13:
@dangowrt 

Thanks, that did the trick!

bananos, 2025-07-15:
@dangowrt I do have the dumps of original device -- but not sure what kind of backup instructions I've followed (it was years ago, before OKD was discovered)

Here's what I got:
```
ls -lh mt*
-rw-r--r--  1 bananos  staff   512K Feb 26  2024 mtd0
-rw-r--r--  1 bananos  staff   4.0M Feb 26  2024 mtd1
```

```
strings mtd1 | grep _addr=
tee_boot_arg_addr=0x%x
wan_mac_addr=C4:41:1E:F5:29:5A
lan_mac_addr=C4:41:1E:F5:29:5B
wifi_2g_mac_addr=C4:41:1E:F5:29:5C
wifi_5g_mac_addr=C4:41:1E:F5:29:5D
```

What are my options?  Is there any way to extract data to populate `factory` partition?

bananos, 2025-07-15:
@dangowrt Daniel, thanks a lot! 
Just to make sure I've understood what to do next

1) Edit `factory` binary you've just sent me to include correct MACs
2) Do something like the following:
```
ubidetach -d 0
insmod mtd-rw i_want_a_brick=1
mtd write /tmp/factory /dev/mtd2
```

bananos, 2025-07-15:
>>edit: sorry it took me 20 minutes to reply, I was reading your website about the coffee grinders because I was planning to get one as a gift. Sadly that model doesn't seem to be available anywhere any more...

@dangowrt Glad to hear someone actually found it useful after so many years, it's a pity they aren't manufactured anymore :(

Anyway, seems like I shot myself into the foot once again -- my previous attempts were to sysupgrade from 23.05.x to 24.10, which has different UBI layout 🤦  . 

This is what I get now via serial connection:

```
F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0400 0041 [0000]
G0: 1190 0000
T0: 0000 02AD [000F]
Jump to BL

NOTICE:  BL2: v2.4(release):OpenWrt v2021-05-08-d2c75b21-3 (mt7622-snand-1ddr)
NOTICE:  BL2: Built : 02:55:34, Sep  3 2022
NOTICE:  SPI-NAND: W25N01GV (128MB)
ERROR:   BL2: Failed to load image id 3 (-2)

```

Tried to boot it into recovery mode by holding reset, but no luck.

My next steps failed too:

```
./mtk_uartboot -a -p mt7622-ram-1ddr-bl2.bin -f openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip -s /dev/tty.usbserial-1420 && screen /dev/tty.usbserial-1420 115200
mtk_uartboot - 0.1.1
Using serial port: /dev/tty.usbserial-1420
Handshake...

```

never got past `Handshake` after many attempts.  Manual suggests that, the issue is in my serial adapter (I have some cheap noname PL2303HX from aliexpress) It's weird, since it worked before like a charm -- I was able to use U-Boot menu without any problems

Any suggestions?

dangowrt, 2025-07-15:
try power off and power on again the device while in `Handshake...` phase, that helps sometimes if the first attempt doesn't work. Other than that it can only be the serial adapter (but even PL2303 should be fine if you set it to 115200 in `mtk_uartboot`) or one of RX, TX, GND isn't connected well...

dangowrt, 2025-08-20:
Ok, the files in boot_backup hint that you have never been running the v1.1.x installer, but probably just used the initramfs-recovery or sysupgrade file which came with it (which would explain the outcome).
In order to recover the device you need to first use `mtk_uartboot` to load U-Boot (`.fip`) which came with OpenWrt 23.05.x.
Once you are inside U-Boot, enter the command shell and use it to load and write the factory data back onto the flash:
```
ubifsmount boot_backup
ubifsload $loadaddr mtd2
mtd erase spi-nand0 0x1c0000 0x100000
mtd write spi-nand0 $loadaddr 0x1c0000 0x100000
```

Now load the installer file via TFTP. Make sure you really use the installer this time, ie. [`openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb`](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.1.4/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb) (note the word `installer` is part of the filename!). Download the file to your computer and provide it on a TFTP server listening on address 192.168.1.254/24.

```
tftpboot $loadaddr openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
bootm
```

The installer should run now and take of migrating everything to be ready to run OpenWrt 24.10.x.

jssst4, 2025-08-20:
@dangowrt  - Steps followed but getting an error. 
1) Restored 23.5.5 using mtk_uartboot (fully working).

BusyBox v1.36.1 (2024-09-23 12:34:46 UTC) built-in shell (ash)
 -----------------------------------------------------
 OpenWrt 23.05.5, r24106-10cc5fcd00
 -----------------------------------------------------

**root@OpenWrt:/# cat /etc/openwrt_release**
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.5'
DISTRIB_REVISION='r24106-10cc5fcd00'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.5 r24106-10cc5fcd00'
DISTRIB_TAINTS=''

2) Reboot & login using mtk_uartboot and selected "U-Boot console" to enter command shell

**MT7622> ubifsmount boot_backup**
UBIFS error (pid: 1): cannot open "boot_backup", error -22
Error reading superblock on volume 'boot_backup' errno=-22!
ubifsmount - mount UBIFS volume

Usage:
ubifsmount <volume-name>
    - mount 'volume-name' volume

dangowrt, 2025-08-20:
Sorry, try `ubi info layout` (instead of `-l`, that was wrong).

dangowrt, 2025-08-20:
Ok, strange, so UBI is doing fine and the volume `boot_backup` (which you had already mounted under Linux) is detected fine also in U-Boot, but you can't mount it, which is weird.
Let's try to just do the same thing under Linux then. You need to start 23.05.x initramfs recovery image, connect WAN port to a network which gives you Internet via DHCP, then
```
opkg update
opkg install kmod-mtd-rw
insmod mtd-rw i_want_a_brick=1
mount -t ubifs ubi0:boot_backup
mtd write /mnt/mtd2 /dev/mtd2
```

Also make sure at this point to copy the files contained in the `boot_backup` volume (now mounted at `/mnt`) to a safe location (eg. a USB pendrive, or use `scp` to copy them to a computer connected to the router). Loosing them would complicate things a lot.
Do **not** proceed unless you got them backed up.

Once you have made the backup once again use `mtk_uartboot` to start U-Boot of OpenWrt 23.05.x, and provide the [installer image](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v1.1.4/openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb) on a TFTP server at 192.168.1.254/24 on a computer connected to any of the Ethernet ports of the router. Then do

```
tftpboot $loadaddr openwrt-24.10.0-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
bootm
```

You should see the installer running and subsequently you no longer need `mtk_uartboot`, but a recent U-Boot should now start without issues and load the updated 24.10.0 initramfs recovery image. From there you can now sysupgrade the 24.10.x release sysupgrade image and be done.

Prototyped, 2025-09-02:
I ran into very similar issues -- BL2 was at v2.4 and unable to boot, I was unable to use the 1.1.4 installer due to mangled `factory` partition, I did have the `boot_backup` volume in ubi0, u-boot was unable to mount it, but fortunately I was able to mount it in the recovery system on flash that I booted from the u-boot menu (using mtk_uartboot to kick things off), and able to use the `mtd write` command from https://github.com/dangowrt/owrt-ubi-installer/issues/217#issuecomment-3207635105 to recover the `factory` partition. Subsequently the 1.1.4 installer did its job and reworked the partition layout to leave one `bl2` MTD partition and one `ubi` MTD partition.

The `boot_backup` volume now reflects the new partitioning scheme, so I'm happy that I copied it off elsewhere before the installer did its thing. (The new partition layout with 24.10.2 is also backed up separately.)

Thanks very much, @dangowrt!

Prototyped, 2025-09-02:
> [@dangowrt](https://github.com/dangowrt) - No luck, seems I have screwed up royally :-(
> 
> root@OpenWrt:/# opkg install kmod-mtd-rw Installing kmod-mtd-rw (5.15.167+git-20160214-2) to root... Downloading https://downloads.openwrt.org/releases/23.05.5/targets/mediatek/mt7622/packages/kmod-mtd-rw_5.15.167%2bgit-20160214-2_aarch64_cortex-a53.ipk Configuring kmod-mtd-rw. root@OpenWrt:/# insmod mtd-rw i_want_a_brick=1 [ 65.922567] mtd-rw: mtd0: setting writeable flag [ 65.927215] mtd-rw: mtd1: setting writeable flag [ 65.931832] mtd-rw: mtd2: setting writeable flag root@OpenWrt:/# mount -t ubifs ubi0:boot_backup mount: can't find ubi0:boot_backup in /etc/fstab
> 
> Kindly guide way forward. Thanks

What's the output of `ubinfo -a`?